realthunder/FreeCAD_assembly3 #1021: Latest daily build crashes when opening a file which contains imported meshes

invader-zimm, 2022-01-11:
I'm using the latest daily build on Windows, and FreeCAD crashes when I try to open a file saved with the previous build (2021.10.15).
As far as I can tell, it's because the file contains imported meshes. I can open files saved with the previous build without any problems if they don't contain any meshes.

To reproduce the issue:
* open the file from the attached [crash.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7850452/crash.zip)
* FreeCAD crashes 
[image]

luzpaz, 2022-01-12:
Please always add your full FreeCAD info (for posterity).

invader-zimm, 2022-01-12:
Sorry, here it is:
```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

invader-zimm, 2022-01-22:
I tried the build linked in [this comment](https://github.com/realthunder/FreeCAD_assembly3/issues/1024#issuecomment-1015342924) and the crash doesn't occur anymore. I think this can be closed now.

closed 2022-01-22 by invader-zimm