realthunder/FreeCAD_assembly3 #1024: shuts down

hamza00789, 2022-01-12:


bambukouk, 2022-01-12:
It would be useful to confirm whether it affects both daily and stable?

hamza00789, 2022-01-12:
> It would be useful to confirm whether it affects both daily and stable?

only Daily

Rabbit-sk, 2022-01-12:
For information, I tried it and this problem did not occur. 

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

luzpaz, 2022-01-12:
@hamza00789 please use more detailed ticket summaries so bug triage is made easier. Thanks

hamza00789, 2022-01-12:
> @hamza00789 please use more detailed ticket summaries so bug triage is made easier. Thanks

I downloaded it again but it is still crashing. The freecad crashes when I go to the mash design workbench

luzpaz, 2022-01-13:
@hamza00789 can you start FreeCAD from the command line to see what it outputs?

hamza00789, 2022-01-13:
> @hamza00789 can you start FreeCAD from the command line, recreate the crash, copy/paste the output back to this ticket?

I don't understand you. It is constantly crashing. Is there a solution to this problem I am facing? Am I the only one facing this problem?  It crashes in the Daily Branch. It works well in the stable branch. The older version I have is working fine. Why the new version is crashing. Can you help me with that?

bambukouk, 2022-01-13:
people are trying to help you
follow the instructions:

```
start FreeCAD from the command line
recreate the crash
copy/paste the output back to this ticket
```

bambukouk, 2022-01-13:
It hasn't crashed in the vid you posted?
Can you carry on until it crashes? and then copy and paste the messages (not vid) on terminal?

hamza00789, 2022-01-13:
> It hasn't crashed in the vid you posted? Can you carry on until it crashes? and then copy and paste the messages (not vid) on the terminal?

Maybe I don't understand you. Don't anyone else has this problem and only me 

https://user-images.githubusercontent.com/92737149/149358137-83dd1eec-aa22-4d26-97cb-861b9b20fbcd.mp4

bambukouk, 2022-01-13:
`start FreeCAD from the command line `- done

`recreate the crash` - done

`copy/paste the output back to this ticket` - **_not done_**

Can you go back to the terminal and copy the messages when your FreeCAD crashed?

There is also: https://github.com/realthunder/FreeCAD_assembly3/issues/1021#issue-1099713790

hamza00789, 2022-01-13:
> `start FreeCAD from the command line `- done
> 
> `recreate the crash` - done
> 
> `copy/paste the output back to this ticket` - **_not done_**
> 
> Can you go back to the terminal and copy the messages when your FreeCAD crashed?
> 
> There is also: [#1021 (comment)](https://github.com/realthunder/FreeCAD_assembly3/issues/1021#issue-1099713790) 

No message arrives at the terminal when FreeCad crashes.

bambukouk, 2022-01-13:
`15:59:54  <App> GeoFeatureGroupExtension.cpp(577): crash#Group001.Group(links to crash#main_body_side_plate001_solid) is out of scope:crash#Part002 vs. <global>
`

hamza00789, 2022-01-13:
> when opening file attached to the linked issue, doesn't crash on Linux but gives the following error message:
> 
> `15:59:54 <App> GeoFeatureGroupExtension.cpp(577): crash#Group001.Group(links to crash#main_body_side_plate001_solid) is out of scope:crash#Part002 vs. <global> ` can you attach yours? so that I can try?            

The attached file of the associated problem may be an error when opening in Linux but freecad crashes when opening in windows

bambukouk, 2022-01-13:
and the knowledge of this (if it is indeed the case) helps in understanding what might be the problem....

hamza00789, 2022-01-13:
> and the knowledge of this (if it is indeed the case) helps in understanding what might be the problem...

https://user-images.githubusercontent.com/92737149/149384561-449a2c19-65cb-477e-8367-fc86ac560680.mp4

hamza00789, 2022-01-13:
> and the knowledge of this (if it is indeed the case) helps in understanding what might be the problem...

Will there be a solution to the problem?

bambukouk, 2022-01-13:
I am not suggesting that you need to prove anything...
what I am trying to "prove" is whether it happens in windows only
and so far it seems to be the case based on file that someone else shared
if you let us have a copy of your file I will test it on Linux

hamza00789, 2022-01-13:
> I am not suggesting that you need to prove anything... what I am trying to "prove" is whether it happens in windows only and so far it seems to be the case based on a file that someone else shared if you let us have a copy of your file I will test it on Linux 

old freecad version[OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)]  

File....
[Nife.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7865245/Nife.zip)                
This is an old free cad version file in which STL was imported

Freecad crashes when I try to open this file in a newer version of freecad. 
 
the new version [OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)]

hamza00789, 2022-01-13:
> Opens perfectly fine without even any error messages
> 
> [image]
> 
> ```
> OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
> Word size of FreeCAD: 64-bit
> Version: 2022.111.30764 (Git) AppImage
> Build type: Release
> Branch: LinkDaily
> Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
> Python version: 3.9.9
> Qt version: 5.12.9
> Coin version: 4.0.1
> OCC version: 7.5.3
> Locale: English/United Kingdom (en_GB)
> ```                

Should I change the operating system?   Or will there be a solution?   Remember I have an old version in which the file opens properly. Would you like to tell me something about that? If I'm not bothering you

bambukouk, 2022-01-13:
I am just a user, like yourself, so unlikely to be able to give you an authoritative answer,
however... I am convinced that if a solution is possible it will be implemented.

Remember, that this version of FreeCAD  is experimental, and if you need something more stable and with broader support you might wish to consider master branch?

hamza00789, 2022-01-13:
> I am just a user, like yourself, so unlikely to be able to give you an authoritative answer, however... I am convinced that if a solution is possible it will be implemented.
>
> Remember, that this [...]

Well, this is an experimental version. I think something that was happening before should still happen or be better. But whether I like it or not, nothing is going to happen. Give  the link to Master Branch

luzpaz, 2022-01-16:
You can start freecad from the command line and use one of these flags:
```
Configuration:
  -l [ --write-log ]        Writes FreeCAD.log to the user directory.
  --log-file arg            Unlike --write-log this allows logging to an 
                            arbitrary file
```

hamza00789, 2022-01-16:
> 

[FreeCAD.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7878146/FreeCAD.zip)

realthunder, 2022-01-17:
The problem is caused by `netgen` (a library used by `flatmesh` and other meshing library used by FreeCAD) default build setting that auto enables extended CPU instruction sets that are avaiable on the build machine. The resulting binary cannot run on older CPU that does not have the instruction set. The relavent fix is mentioned in [this](https://forum.freecadweb.org/viewtopic.php?style=1&p=485717#p485568) post. I'll contact the person who built the `netgen` library to fix this.

realthunder, 2022-01-18:
@hamza00789 Can you please try [this](https://drive.google.com/file/d/1aVGnKlVCGe7Sr6ZxOMMNdRo58bmiEgPv/view?usp=sharing) and report back if it works.

hamza00789, 2022-01-18:
> @hamza00789 Can you please try [this](https://drive.google.com/file/d/1aVGnKlVCGe7Sr6ZxOMMNdRo58bmiEgPv/view?usp=sharing) and report back if it works.

Yes, it works fine now. Should I use this version now?

hamza00789, 2022-01-18:
thanks to solving

closed 2022-01-18 by hamza00789

RayPS, 2022-02-03:
This issue also happened to me on macOS (Latest Monterey).

I'm using the latest macOS Monterey and `FreeCAD-asm3-Stable-BigSur-OSX-Py3-Qt5-20220111-x86_64.dmg`

#### Reproduce steps:
1. Open FreeCAD
2. Switch to MeshDesign work bench
3. Crash

#### Crash Report:
<details>
  <summary>Click to expand</summary>
<pre>
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               FreeCADLink [71495]
Path:                  /Applications/FreeCAD Link Branch.app/Contents/MacOS/FreeCADLink
Identifier:            
Version:                (2022)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-02-03 20:17:07.9877 +0800
OS Version:            macOS 12.0.1 (21A559)
Report Version:        12
Anonymous UUID:        A134D343-5823-9779-BD9D-DAD95C5FABD6

Time Awake Since Boot: 490000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [71495]

Kernel Triage:
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libSMDS.dylib                 	       0x197858994 _GLOBAL__sub_I_SMDS_Downward.cpp + 4
1   dyld                          	       0x2026ecb49 invocation function for block in dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 182
2   dyld                          	       0x2027130ff invocation function for block in dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 129
3   dyld                          	       0x20270a893 invocation function for block in dyld3::MachOFile::forEachSection(void (dyld3::MachOFile::SectionInfo const&, bool, bool&) block_pointer) const + 566
4   dyld                          	       0x2026d9d91 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 129
5   dyld                          	       0x20270a61b dyld3::MachOFile::forEachSection(void (dyld3::MachOFile::SectionInfo const&, bool, bool&) block_pointer) const + 179
6   dyld                          	       0x202712b30 dyld3::MachOAnalyzer::forEachInitializerPointerSection(Diagnostics&, void (unsigned int, unsigned int, unsigned char const*, bool&) block_pointer) const + 118
7   dyld                          	       0x202712da2 dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 386
8   dyld                          	       0x2026eca7c dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 144
9   dyld                          	       0x2026ecc08 dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 178
10  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
11  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
12  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
13  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
14  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
15  dyld                          	       0x2026ecbeb dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 149
16  dyld                          	       0x2026eccac dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 108
17  dyld                          	       0x2026fb9b5 dyld4::APIs::dlopen_from(char const*, int, void*) + 563
18  Python                        	       0x10f400d50 _imp_create_dynamic + 1309
[... 628 lines cut ...]
USB Device: USB31Bus
USB Device: USB 2.0 Hub
USB Device: Generic Billboard Device
USB Device: USB31Bus
USB Device: USB30Bus
USB Device: Game Capture HD60 S+
USB Device: USB2.0 Hub
USB Device: USB Condenser Microphone
USB Device: USB Receiver
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

</pre>
</details>

#### Run in terminal result:
```
/Applications/FreeCAD\ Link\ Branch\ \(BigSur\).app/Contents/MacOS/FreeCADLink -l
FreeCAD 0.19, Libs: 2022.111R28429 (Git)
0.892193 <asm3.main> init_gui.py(14): no solver backend found
[1]    72454 illegal hardware instruction  /Applications/FreeCAD\ Link\ Branch\ \(BigSur\).app/Contents/MacOS/FreeCADLin
```

#### Log file:
[FreeCAD.log](https://github.com/realthunder/FreeCAD_assembly3/files/7994533/FreeCAD.log)