realthunder/FreeCAD_assembly3 #1025: [bug], daily 2022.111 Polyline tangent, perpendicular etc styles with "M" not working

Miksu80, 2022-01-12:
As description says, cycling modes with "M" not working at all.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/Finland (en_FI)

Rabbit-sk, 2022-01-12:
I noticed that too. It works with `G, M`. Some shortcuts have changed in Sketcher.

QuakeString, 2022-01-14:
yes, you can hover over the icon and it will show the key combination. 
this is the new thing. before it was M now it is G and M.

Miksu80, 2022-01-14:
New combo is not good with polyline because every time you're switching mode, you've got to press G, M -> from any direction polyline to perpendicular circle you've got to press: G,M,G,M,G,M,G,M,G,M from previous M,M,M,M,M. It'd be just fine if the polyline -tool selection would be G,M but to cycle the modes when in polyline it'd require only "M".

QuakeString, 2022-01-14:
> New combo is not good with polyline because every time you're switching mode, you've got to press G, M -> from any direction polyline to perpendicular circle you've got to press: G,M,G,M,G,M,G,M,G,M [...]

Totally agreed. I believe dev. will take a notice.

Rabbit-sk, 2022-01-14:
> It'd be just fine if the polyline -tool selection would be G,M but to cycle the modes when in polyline it'd require only "M".

I agree with you. An important part of this proposal is: 
> to cycle the modes when in polyline it'd require only "M"

This is because there is a shortcut `M, M` that works with preselection, to display some possible actions with a given object. 
But when drawing a polyline, it is obvious that the user wants cycling between polyline modes, single pressing `M` in this mode should have higher priority than the shortcut `M, M`.

So the proposal we came up with is as follows: 

- Keep shortcut for polyline tool selection `G, M` as it is now.
- Due to the abbreviation `M, M` evaluate whether the polyline is the currently selected tool. 
- Change shortcut for cycling polyline modes from `G, M` to `M`.

realthunder, 2022-02-28:
For those who are interested, the above commit introduces dynamic shortcut conflict resolving in FreeCAD. It can now perform longest key sequence match instead of Qt's default shortest match. In other word, Shortcut `M` and `M, M` can now coexist without any problem. I'll see if I can submit this to upstream and suggest Sketch to revert back the original shortcuts.

luzpaz, 2022-03-02:
JFYI, upstream Feature Freeze is March 4th