realthunder/FreeCAD_assembly3 #1028: update exported sketch geometry

kevenwyld, 2022-01-13:
Forgive me if this is already a feature but is there a way to update exported sketch geometry without creating a new export and deleting the old one?  If not I think this would be a handy feature.

```
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

hamza00789, 2022-01-14:
> Forgive me if this is already a feature but is there a way to update exported sketch geometry without creating a new export and deleting the old one? If not I think this would be a handy feature. [...]

No need to delete old geometry, you can update it 

https://user-images.githubusercontent.com/92737149/149570265-f92a9c27-4be6-4713-a360-f4c9ef80146e.mp4

kevenwyld, 2022-01-14:
Thanks, I'll give that a try.  Just to make sure I understand what you did in the video, are you pasting a new set of edges into the `Base Refs` attribute?

If that's the case, and this is the only way to do it, I think a button visible when opening an exported sketch would be better.  The sketcher already highlights the exported geometry when you double click on the exported sketch (see video).  So selecting on additional line and clicking an "update export002" button would be much more convenient.

https://user-images.githubusercontent.com/11936535/149572251-36232739-90a6-4a0f-b391-528385f82ffd.mp4

kevenwyld, 2022-01-14:
Nevermind, I found a more convenient way, you can select by clicking on the edge in the 3d view.  That works great for me so although a button in the sketcher would be more convenient for complex sketches, I think this is sufficient.

https://user-images.githubusercontent.com/11936535/149572662-5ab98506-1abc-45e3-93cd-7b894bcb2cbf.mp4

hamza00789, 2022-01-14:
Once the problem is resolved, it is best to close it

kevenwyld, 2022-01-14:
Since the focus of the fork on which this release is based seems in large part to be a more user friendly UI, I'd like @realthunder to weigh in on the potential of an update export button.  Double clicking on the export to open the sketch which then presents you with the exported geometry highlighted but having no way to update it from the sketcher seems unintuitive so I wouldn't be surprised if it's already on his radar.

hamza00789, 2022-01-15:
> Since the focus of the fork on which this release is based seems in large part to be a more user friendly UI, I'd like @realthunder to weigh in on the potential of an update export button. Double [...]

The fact is that freecad has no way to update the export from inside the sketch. I can't say more because I'm just a freecad user. 
Instead of export geometry, there is a command named output feature in the CATIA. The output feature in CATIA v5 can be updated from within the sketcher. You may be requesting this feature.                         

https://www.youtube.com/watch?v=YOOh5_T7HmU

hamza00789, 2022-01-16:
> Since the focus of the fork on which this release is based seems in large part to be a more user-friendly UI, I'd like @realthunder to weigh in on the potential of an update export button. [...]

I found that the export geometry can also be updated from within the sketch. But no dialog box or anything like that opens for it. same way as in 3D      

https://user-images.githubusercontent.com/92737149/149659698-3c9b7b3d-7311-4ce0-bde4-14e958d5d55b.mp4

kevenwyld, 2022-01-16:
Thanks, actually that is a much better UX than doing it from the standard 3d view.  You get multiple lines highlighted simultaneously without holding the ctrl key.  Thanks for pointing that out @hamza00789! Also the lines which are already part of the export are already highlighted when double clicking on the export in the tree (which opens the sketcher)

https://user-images.githubusercontent.com/11936535/149672778-c4fa7534-e875-44e7-8bf6-5ad01f862a52.mp4

There seem to be multiple workarounds presented here in this github issue, but none of them are particularly obvious and don't present themselves as a "feature" to update sketches using the sketch export functionality.

kevenwyld, 2022-06-05:
@luzpaz or @realthunder  would you mind moving this to https://github.com/realthunder/FreeCAD ?

Thanks in advance!