realthunder/FreeCAD_assembly3 #1029: shotcut does not work in Sketcher WB.

QuakeString, 2022-01-13:
constrain distance shortcut [K, D] some time works most of the time doesn't.

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

wohltat, 2022-01-14:
Could not find a case when it did not work.

By the way, when and why have the shortcut keys been changed?

QuakeString, 2022-01-14:
Sorry, My bad, I discovered that now shortcut work in key combination. before it was just one key, now it is combination now.

closed 2022-01-14 by QuakeString