realthunder/FreeCAD_assembly3 #1030: TechDraw drawings suffer from TNP when freezing or unfreezing (sub)assemblies, even if no geometry changed

ToniTen, 2022-01-13:
Say you've doing a complex assembly, and you follow best practices: 

- make small sub-assemblies
- freeze them
- build bigger sub-assembles composed of those sub assemblies
- freeze those sub-assemblies
- eventually build the final assembly
- freeze it
- create TechDraw drawings of the final assembly (for catalogs, to send to customers, to apply for projects, etc), including section views, etc

The problem is, even if you change nothing, if you unfreeze the final assembly and/or the sub assemblies, your TechDraw drawings are going to suffer from the TNP, and all dimensions will need to be redone.

I understand that the TNP is a very complex issue, specially when dealing with TechDraw 2D projections that are the result of projecting the model in 2D and then doing HLR on it. However, in this case, no geometry changed, just changing it to be frozen or not.

ToniTen, 2022-01-13:
On a related note, is it possible to extend the benefits of your toponaming scheme to TechDraw drawings? I had some discussion with WandererFan on the forums and had an idea that might make TechDraw drawings more resilient to TNP issues. I don't have more than a superficial understanding of the internals of FreeCAD (though I've read how your toponaming algorythm works), and WandererFan was right to point that it would make an already slow process slower, but better slower but stable than still slow but extremely fragile.
This was my idea on how to at least make it more resilient: https://forum.freecadweb.org/viewtopic.php?p=560568#p560568

luzpaz, 2022-04-13:
Is this Assembly3 related or Link Branch related?

realthunder, 2022-04-13:
The OP is more asm3 related I believe. @ToniTen can you please provide a file showing the problem. I tried your steps but didn't seem to have any problem.

ToniTen, 2022-04-13:
> The OP is more asm3 related I believe. @ToniTen can you please provide a file showing the problem. I tried your steps but didn't seem to have any problem.

Last time that happened was a while ago, and I've been careful since then. I will try to create a file illustrating the problem that I can share. It might not happen any more for all I know. I also know that this has happened to me when dealing with large assemblies, usually partly composed of imported steps (so the geometry isn't always in a good shape)