realthunder/FreeCAD_assembly3 #1031: Cannot start the newest version of FreeCAD (Ubuntu 20.04)

Ejoz, 2022-01-13:
Hello @realthunder, first of all, thank you for the amazing work. I use your version of FreeCAD for a while now, and I love it.

I try since yesterday to install the newest version (20220111), and I keep getting this error:

```
/tmp/.mount_FreeCAVRoLW7
/tmp/.mount_FreeCAVRoLW7/usr/bin/FreeCADLink: error while loading shared libraries: libOpenGL.so.0: cannot open shared object file: No such file or directory
```

I never had this before with your versions, so I'm not sure what to do. For now I just reinstalled the old version and it works perfectly, so it's not a major issue for me, but I thought you'd like to know.

Have a nice day, and thanks again for your work.

bambukouk, 2022-01-13:
there are at least two posts on the subject 
install libOpenGL.so.0

Ejoz, 2022-01-13:
Thank you, I wasn't sure, and didn't see the posts, sorry.

bambukouk, 2022-01-13:
no problem at all :+1: