realthunder/FreeCAD_assembly3 #1038: Panels layout problem

Rabbit-sk, 2022-01-15:
Hi RealThunder!

Sometimes it happens that the layout of the panels breaks down.
Example:
[image]
I unintentionally changed the layout of the panels, and subsequently, even after an hour of dragging and dropping the panels, I did not achieve the desired result:
[image]
In previous builds, I somehow achieved this desired result. 
The solution would be to delete or rename the configuration file, but I don't want to set that many settings again. 

It would be nice if the layout of the panels could be saved in a separate file and loaded later if necessary. 

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-15:
By the way, is opening previously closed panels the desired result of pressing the shortcut `F4`?
[image]

https://user-images.githubusercontent.com/74265320/149618021-9586e9ea-e075-45b3-b993-736cbf0bb6be.mp4

Rabbit-sk, 2022-01-15:
Ideally, it could be like this:
[image]

Rabbit-sk, 2022-01-15:
UPDATE:
> The solution would be to delete or rename the configuration file, but I don't want to set that many settings again.

It didn't solve the problem.

realthunder, 2022-01-15:
Yeah, I didn't really implement tabifying dockable panels using drag and drop. You can temporarily work around this by doing the following 'not so natural' sequence.
* Toggle `F4` to make sure no dockable widget is in overlay mode.
* Drag and drop to rearrage the dockable widgets in order of your preference
* hold `SHIFT`
* Press mouse left button to start dragging the title bar of the top most docked widget. Please note, only the top most title bar works.
* Release the mouse button, and you can now drag to tablify (i.e. group the tabs) of the docking widget.

https://user-images.githubusercontent.com/1207888/149623671-3f29f034-f6bf-495b-87a7-beb3c45f21b8.mp4

bambukouk, 2022-01-15:
> ... The solution would be to delete or rename the configuration file, but I don't want to set that many settings again.

the solution would be to have a backup copy of stable/latest config file :+1: 
yes, panels layout is a bit of PITA, as it is at the moment

Rabbit-sk, 2022-01-15:
> You can temporarily work around this by doing the following 'not so natural' sequence.

Thank you very much @realthunder !

Rabbit-sk, 2022-01-15:
I've changed the name of the issue and will close it later when you implement this change.

closed 2022-01-16 by realthunder