realthunder/FreeCAD_assembly3 #1039: [feature request] use camera orientation to set sketch Map Reversed / Flip sides

ohsix, 2022-01-15:
hi

maybe freecad already does this but gets it' backwards from my expectation.

when i'm looking at a face i (or plane) i want to put a feature on, i want the pad to come away from that face, towards the camera

right now it basically gets it exactly wrong nearly every time and i'm not sure what the actual rules are

thank you :)

Rabbit-sk, 2022-01-15:
Doesn't it work like this for you? 

https://user-images.githubusercontent.com/74265320/149618692-5dbe0293-f13e-4922-b90d-3645718daf83.mp4

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

ohsix, 2022-01-16:
it doesn't orient the sketch correctly every time. here's me doing it backwards to demonstrate it's not using the camera orientation to decide

https://user-images.githubusercontent.com/3180944/149653429-36f4e98e-7cd4-4677-82c7-aeabeb3321e9.mp4

i do not normally hold shift to pick a backface like this, generally i'm looking at the face in the orientation i want the sketch to come from and sometimes the existing freecad facing logic gets it wrong

and in models where i have manually gone in and changed Map Reversed, those sketches are now also upside down, complicating the mess

realthunder, 2022-01-16:
I have just added a new action to orient sketch to the oppsite side. BTW, you can also do this by clicking that little cicular dot button near the Navigation Cube.

[image]

hamza00789, 2022-01-16:
> I have just added a new action to orient sketch to the oppsite side. BTW, you can also do this by clicking that little cicular dot button near the Navigation Cube.
> 
> [image]

Is this problem similar to this?    
Will this solve my problem to some extent? 
#908