realthunder/FreeCAD_assembly3 #1042: Error on selecting WPProxy as working plane "Object Without Part.Shape eeometry:' WPProxy'.

QuakeString, 2022-01-15:
When ever I'm trying to select WPProxy as current working plane it showing this error
[image]
This error shows when finish BIM Project Setup.
[image]
My system detail.
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

QuakeString, 2022-01-17:
I've tried freeCAD main branch .19.3 and it works properly.
[image]

luzpaz, 2022-01-17:
Can you reproduce on latest upstream 0.20dev (0.19.3 is irrelevant unless you're showing a regression and still need to test the latest v0.20dev nonetheless to understand if it is a regression)

QuakeString, 2022-01-17:
> Can you reproduce on latest upstream 0.20dev (0.19.3 is irrelevant unless you're showing a regression and still need to test the latest v0.20dev nonetheless to understand if it is a regression)

There is no issue in FreeCAD main Daily dev branch. 

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27078 (Git) AppImage
Build type: Release
Branch: (HEAD detached at bcdd926)
Hash: bcdd926686f2245fe60baface1cce1ab3dff7a5b
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

closed 2022-01-18 by realthunder

wolfgangr, 2024-02-27:
I get the same message when I try to follow the tutorial
https://wiki.freecad.org/Manual:Traditional_2D_drafting/de

beginners mistake?
regression bug?

Steps to reproduce:

- drafting 2D as given in the tutorial
- extruding walls, as given in the tutorial, using "part" workbench
- try to change frame of reference

```
20:41:57  GuiCommand: Trimex
20:41:57  Wählen Sie Objekt(e) zum verkürzen/verlängern
20:42:01  Wähle eine Fläche, 3 Punkte oder einen WP Proxy um die Zeichenebene zu definieren
20:42:18  ----------------
20:42:18  GuiCommand: Trimex
20:42:18  Diese Objekte überschneiden sich nicht.
20:44:09  Object without Part.Shape geometry:'Gruppe003'
20:44:10  Object without Part.Shape geometry:'Gruppe003'
20:44:13  Object without Part.Shape geometry:'Gruppe003'
20:44:42  Object without Part.Shape geometry:'Gruppe003'
20:44:42  Object without Part.Shape geometry:'Gruppe003'
```

[code]
OS: Debian GNU/Linux 10 (buster) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.12
[/code]