realthunder/FreeCAD_assembly3 #1046: Hint not displayed on toolbar icons in Sketcher

Rabbit-sk, 2022-01-16:
Hi RealThunder!

If the cursor is over the toolbar icon in Sketcher, the hint is not displayed correctly.

https://user-images.githubusercontent.com/74265320/149653129-7353d051-0c40-40a3-b02d-c756638a910a.mp4

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-16:
UPDATE:

https://user-images.githubusercontent.com/74265320/149654373-dd22fb05-1799-47a3-81e6-8ca615d79835.mp4

https://user-images.githubusercontent.com/74265320/149654405-1ad81dd0-03eb-4fc5-89d1-326c809528bc.mp4

Rabbit-sk, 2022-01-16:
Could it be related to this?

https://user-images.githubusercontent.com/74265320/149654481-e5d11a81-4e5f-48bd-82bb-31c27a6aa6a5.mp4

realthunder, 2022-01-16:
Can you please go to `Tools -> Edit parameter` and export your application settings and posted here?

realthunder, 2022-01-16:
Please export the parameter while this problem is happening.

Rabbit-sk, 2022-01-16:
[link.user.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7876802/link.user.zip)

I also noticed that only a small area works to drag the toolbar that is next to the 3D view. You can notice this in the second video of this issue at 00:15.
[image]

Rabbit-sk, 2022-01-16:
Sorry, this file is what you want:
[Parameters.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7876821/Parameters.zip)

Rabbit-sk, 2022-01-16:
I exported the previous file by first hovering over icons in Sketcher, then minimizing FreeCAD, looking at your message on how to export the settings, and then exporting them. I exported this second file as soon as I hover over the Sketcher icon. Differences were found in the content of these files, so I am attaching this additional file as well.
[Parameters_2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7876874/Parameters_2.zip)

Rabbit-sk, 2022-01-16:
UPDATE:
Clicking on such an icon in the middle and not on the margin has no effect.

closed 2022-01-19 by realthunder

Rabbit-sk, 2022-02-10:
I found that this problem also appeared on the toolbar in Ddraft workbench.