realthunder/FreeCAD_assembly3 #1047: Corner render issue

Rabbit-sk, 2022-01-16:
Hi RealThunder!

The corner is sometimes rendered like this:

https://user-images.githubusercontent.com/74265320/149660121-41171767-41a3-4c67-a6b1-b86acd28a886.mp4

[Parameters_corner_render_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7876941/Parameters_corner_render_issue.zip)
Password as usually: RealThunder_came__RealThunder_saw__RealThunder_fixed

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

bambukouk, 2022-01-16:
wouldn't be anything to do with this:
https://forum.freecadweb.org/viewtopic.php?f=3&t=64833

hamza00789, 2022-01-16:
> 
They look like a bevel macro. 
And surprisingly, the bevel macro doesn't work.  
#988

bambukouk, 2022-01-17:
[image]
I have tried unzipping the file using password provided (why do we need password?) 
Anybody else managed to open the file?

realthunder, 2022-01-17:
This is long time issue. It has to do with auto view clipping function in Coin3D library. Just click `Fit View (V, F)` to see the full model.

bambukouk, 2022-01-17:
As explained in the thread I have linked....

Rabbit-sk, 2022-01-17:
> I have tried unzipping the file using password provided (why do we need password?)

We don't need a password, especially if it's listed just below the file. It was just for fun and some form of praise for RealThunder. But I admit that praise can be done in another form.

> Anybody else managed to open the file?

I managed to unpack it with a WinRAR with which I created a zip archive and also with 7zip (also on Windows) and with 
[Engrampa](https://mate-desktop.org/) on Linux Mint.

I'm sorry that the password-protected zip archive caused you the inconvenience. If you are still interested in opening the file, here is a zip archive without a password:
[Parameters_corner_render_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7881921/Parameters_corner_render_issue.zip)

bambukouk, 2022-01-17:
Understand what you were doing with the password - I share your sentiment about Realthunder's work.
No, it's not needed any more - we have confirmation of the old bug that I was also suggesting.
BTW I was expecting project file 
(where I would have been able to easily confirm that `view>fit all` solves the problem)
rather than parameter file, but that's another story...

Rabbit-sk, 2022-01-17:
> This is long time issue. It has to do with auto view clipping function in Coin3D library. Just click `Fit View (V, F)` to see the full model.

Since this is a bug that is not in FreeCAD, and has been known for a long time, I will close this issue.

PS: I apologize to everyone for the duplicate issue. I'm testing the FreeCAD Link branch to help improve it. I thought this issue might have something to do with the experimental renderer.

closed 2022-01-17 by Rabbit-sk