realthunder/FreeCAD_assembly3 #1050: Object is not closed after thickness application

Rabbit-sk, 2022-01-16:
Hi RealThunder!

Maybe it's related to  #946 

https://user-images.githubusercontent.com/74265320/149667715-5dbe883d-7d7d-425d-aa99-e5c81c5386e2.mp4

[Thickness.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7877328/Thickness.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

hamza00789, 2022-01-16:
Do you need a temporary solution?

Rabbit-sk, 2022-01-16:
I was just trying some things as part of testing. But I wonder what the temporary solution is?

hamza00789, 2022-01-16:
> I was just trying some things as part of testing. But I wonder what the temporary solution is?

[Thickness 01.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7877496/Thickness.01.zip)      

temporary solution

Rabbit-sk, 2022-01-16:
Thank you. Just for information, this is how I modified it. 
[image]

hamza00789, 2022-01-16:
> Thank you. Just for information, this is how I modified it. [image]

I would like to get some information from you regarding this issue #1024

hamza00789, 2022-01-16:
> 
If you want to provide information you can come to the problem and say what you need to know.

hamza00789, 2022-02-23:
Maybe related #946

realthunder, 2022-03-28:
This is not really a bug. The thickness operation by definition will try to shell the given solid which is an already shelled thin wall, hence the failure. Apart from using multiple bodies, you can also simplify the job by using multiple solids in the same body, as shown below.

https://user-images.githubusercontent.com/1207888/160406635-0a6f0c19-ae42-442a-ac7f-bee52641ee36.mp4