realthunder/FreeCAD_assembly3 #1051: Model breaks with 2021111 release upon parameter change

GinTonic99, 2022-01-16:
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Austria (de_AT)

The attached parametrized model works with the previous release when a parameter is modified ( not perfectly, but with one or two constraints disabled and re-enabled again it is repaired) but completely breaks with the current release with numerous broken elements.

Attached the model Shoji_01 and two scren recordings shwoing the behaviour with the current and the previous releases of linkdaily.

For the moment I have stopped using the current release and went back to the previous release.

[Shoji_01.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7877286/Shoji_01.zip)

https://user-images.githubusercontent.com/34986229/149671124-b2b4d630-c072-441c-b52b-d9729d00cc9f.mp4

https://user-images.githubusercontent.com/34986229/149671353-8623fe44-7f66-4e29-905b-317b50dc597b.mp4

HaasNL2, 2022-01-20:
I have also noticed more rebuild issues than previous.

GinTonic99, 2022-01-20:
Hi realthunder,

thank you for attending the issue. I can confirm that with the updated asm3 from the git repo and upon following your instructions, the model behaves like before with the current release.
thanks again
gin

closed 2022-01-20 by GinTonic99