realthunder/FreeCAD_assembly3 #1055: Assembly 3 - Attachment Tool doesn't seem to work anymore

jx2014, 2022-01-18:
Steps to create issue:
1. Create a body using Part Design workbench.
2. Create a screw hole (set to ISO Metrics and size M4) like so:
3. [image]
4. Use `Fasterners` workbench to import a m4 screw like so:
5. [image]
6. The `Tree view` now looks like this:
7. [image]
8. In `Assembly 3` workbench, create a a new assembly.
9. Use `Link actions` - `Make link` to link the Body and M4 screw to parts.
10. Use `Axial move part (A, A)` to move the M4 screw away from the body a little bit.
11. The body and screw shall look like this:
12. [image]
13. The `Assembly` Tree view should look like this:
14. [image]
15. Select the a feature of the screw hole, and a feature of the m4 screw (holding on Ctrl key):
16. [image]
17. Click on `Add a attachment constraint`.
18. [image]
19. The expectation is that the M4 screw will attach to the screw hole as shown in this youtube video (at 16:00):
20. https://youtu.be/RlKebJtwNCc?t=959
21. But it the constraints appear to do nothing at all.
22. model file is attached here: [2.zip](https://github.com/realthunder/FreeCAD/files/6837400/2.zip)

occurs with 2021.7.17 version:

`OS: Windows 10 Version 2009`
`Word size of OS: 64-bit`
`Word size of FreeCAD: 64-bit`
`Version: 2021.717.24301 +3627 (Git)`
`Build type: Release`
`Branch: LinkStage3`
`Hash: f63a5ea7b459872a3798027df0ff25e776524859`
`Python version: 3.8.6+`
`Qt version: 5.15.2`
`Coin version: 4.0.1`
`OCC version: 7.5.0`
`Locale: English/United States (en_US)`

also occurs with 2021.6.14 version:

`OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3584 (Git)
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)`

realthunder, 2022-01-18:
Can you please try with the latest image? Seems working fine here.

realthunder, 2022-01-18:
BTW, you need to reactivate the solver by disable that attachment and re-enable it. Not sure how it got into this 'stale' state.

luzpaz, 2022-07-01:
@jx2014 ping

realthunder, 2022-07-07:
Can you please try the latest release? Also, make sure you've enabled `Auto solve` option.

[image]

Without auto solve option, you can also manually click `Solve` or `Quick solve` button.

[image]

jx2014, 2022-07-07:
I verified it. 
It is now working.
This ticket can be closed.

Thank you @realthunder 

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.704.28821 (Git)
Build type: Release
Branch: LinkStage3
Hash: c83a0368027dcdb972bfb3e0a00fbf9c26536fcd
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

```