realthunder/FreeCAD_assembly3 #1056: install assembly solver

bambukouk, 2022-01-18:
I have now tested it on totally clean/virgin installation (both OS and FreeCAD)
Results below (apart from the fact that FreeCAD got into some loop and couldn't stop it from trying to install...)
In the end after shutting PC down and restarting  all is fine and solver is installed despite dire warnings previously.
I know it has been reported before but I wanted to try it on totally new installation just in case some of what I did previously might have caused problem?

[image]
[image]

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

```

luzpaz, 2022-01-22:
Any traction on this?

bambukouk, 2022-01-22:
sorry, can you please elaborate?
not sure I follow your question
not very good at git speak :-)

luzpaz, 2022-01-22:
'traction' as in 'progress' :wink:  
Just wondering if @realthunder has addressed this yet?

bambukouk, 2022-01-22:
Thank you 👍 
no, he didn't as far as I know?
It's a bit of "non-issue" because we know that despite warnings the update actually happens, but...
might confuse some (many?)

realthunder, 2022-01-23:
Yeah, the problem should be solved in 361fde2e67bdcf3fc485669e832e7bfe26c90ed3. I'll do another minor release soon.

closed 2022-01-23 by realthunder