realthunder/FreeCAD_assembly3 #1059: AppImage update appears broken

VorpalBlade, 2022-01-19:
Hi,

Not sure if this is a LinkStage3 or upstream issue, but the AppImage update function (so on Linux) from the settings screen does not work.

It appears to correctly download only the changes (so reasonably quick) but then stalls near the end:
[image]

And eventually (a couple of minutes) fails at the very end with the following dialog box:
[image]

Note the stall is always at the same point (99%). Downloading the full AppImage from the github releases page in a browser works fine (but obviously takes longer, and might be an issue for anyone who has a data cap).

I have tested this on two systems and the outcome is the same:
* An Arch Linux desktop
* An Ubuntu 20.04 LTS laptop

System info (before the update) for the Arch Linux desktop (as that is where I am now):
```
OS: Arch Linux (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

realthunder, 2022-01-19:
It seems that the root cause is most likey a `github` rate limit problem. See [here](https://github.com/AppImage/AppImageUpdate/issues/88).

VorpalBlade, 2022-01-19:
Huh, I only skimmed that bug, but that seems more related to update checks, not to the actual download? I may be missing something.

Or are you saying that using zsync creating a lot of separate requests for each segment of the file that needs to be downloaded? I would have assumed that https://developer.mozilla.org/en-US/docs/Web/HTTP/Range_requests would be used (with the multi-range syntax).

luzpaz, 2022-04-29:
This happens from time to time. I've reported the first time back in 2019 https://github.com/AppImage/AppImageUpdate/issues/128

closed 2022-04-29 by luzpaz