realthunder/FreeCAD_assembly3 #1060: Any keybinding that is "Clear"ed and not rebound resets (and then overlaps with others) when restarting

brandonbeaumia, 2022-01-20:
2021.1015 (a little scared to update just yet), but haven't seen anything regarding this in the 2022 release notes.

When attempting to clear the keyboard shortcut of a function in order to make room for it somewhere else, it all seems to work as intended. However, after restarting the app, any binding that was "Cleared" is reset back to its default shortcut, which creates overlaps with whatever you bound in its place.

For example, on a completely fresh install, set Create Line to the F key. It will warn about overwriting Fillet. Accept the overwrite and Assign F to Create Line. Confirm the binding works. Restart FreeCAD and attempt to use the F shortcut again to create a line. It will fail, because F is now bound to both actions.

Is this something to do with how the shortcuts are stored in the XML Parameters? Perhaps a "cleared" shortcut is not registered as being customized, so the fact that it was cleared isn't stored and then upon reboot it goes back to its baked-in shortcut again? Either way, I've spent the last few days working on my keyboard shortcuts and it's driving me crazy to re-clear multiple shortcuts every time I have to restart the app.

Rabbit-sk, 2022-01-20:
I can confirm that.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

StephenBrough, 2022-02-23:
Seeing the same thing. Thought all my keyboard shortcuts randomly stopped working.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

realthunder, 2022-02-24:
@brandonbeaumia Can you please elaborate the problem you have? I can't reproduce it. The default sketch shortcut has been changed from upstream, not by me. For example, `Create line` default shortcut has been changed to `G, L`, and `Create fillet` is changed to `G, F, F`. I tried to manually changed `Create line` to `F`, and seems to be working after restart.

brandonbeaumia, 2022-02-24:
Bad example since the defaults have been updated. I can also confirm that my original report is actually incomplete/incorrect. Here's how to reproduce since the update:

Reset all shortcuts (if not already the defaults) and relaunch FreeCAD.
Create a new file, switch to Sketcher workbench, create a new empty sketch (just making sure everything initializes all the way)
In Keyboard, set Constrain Angle to K,B. It will ask if you want to overwrite Constrain Block. Press yes and hit Assign.
Confirm that Constrain Block's shortcut is now 'none' and Contrain Angle is now K,B.
Relaunch FreeCAD.
Both functions are now set to K,B causing all sorts of weirdness in Sketcher.

However, if you go to Constrain Block and <Clear> it before assigning K,B to Constrain Angle, everything seems to work as intended. So the real issue is that when you choose to assign a shortcut that is already assigned, the "automatic" clearing doesn't stick.

closed 2022-02-28 by realthunder