realthunder/FreeCAD_assembly3 #1062: Proposal: Complete Sketcher shortcut overhaul (and Part Design if you want it)

brandonbeaumia, 2022-01-20:
Hey, folks. I am a hobbiest UI designer and am hoping to get some discussion going regarding the state of default shortcuts (in Sketcher and Part Design specifically). Realthunder recently updated most of the Sketcher shortcuts so I wanted to pounce now before it became too late. For the last few days, I've been working on refactoring Sketcher and Part Design keybindings into one non-overlapping layout and think I've finally come to something worth sharing. This was over 20 hours of thought, design, iteration, and testing.

Here's the layout proposal: https://docs.google.com/spreadsheets/d/1RNlKXoMiDZfG8x6cy7PFdSOp4zO-26aWJYOChBQq6jg/edit?usp=sharing

The benefits and design philosophy are on the first Sheet, but I'll copy them here as well:

**Coverage**
This layout covers almost every single function of Sketcher and Part Design, plus a couple global pie menus for handy View and Tape Measure functions that aren't bound otherwise. Please note that shortcuts beginning with a / character are being used to work around a bug in FreeCAD in which any shortcut that is "Cleared" but not set to another value does not hold its Cleared setting upon restart (and then overlaps with the new stuff you've bound! Very annoying.)

**Overlaps**
Very few shortcuts ask about being overwritten while binding this layout; it seems it is mostly safe across workbenches, in case you also use the default shortcuts of Arch, Draft, FEM, etc.

**Speed for Frequent Tools**
Sketcher shortcuts need to be as fast as possible due to the frequency of tool changes in this workbench, so they are all **reachable with the left hand from home row**. Sketcher has claimed every single-key shortcut besides the W key (which is then used for global List/Pie Menus). Despite sticking to left hand keys only, most of these ended up making a lot of sense anyway. **Poly-line is also back to being a single-key shortcut.**

**Memorability for Less Frequent Tools**
Most Part Design shortcuts are the first two letters of the main word of the function. Infrequently, a shortcut is the first letter each of a pair of words. If you haven't memorized a shortcut you can probably guess correctly 95% of the time.

 **Easy to Learn / Discover**
The half-dozen or so categories that have a (List/Pie Menu) shortcut listed should be bound and memorized ASAP. This creates a very smooth transition from mouse flinger to keyboard maestro. If you're set up for a List menu, you'll see the shortcut right away, and in a Pie menu you can hover to see the shortcut.

I feel like a strong set of default shortcuts could go a long way towards helping FreeCAD's adoption. If anyone has any input on this setup at all, please let me know. And Realthunder, if there are no "dealbreakers" in here, please please consider updating the defaults. Even if you skip all the Part Design shortcuts (since Pie Menus do a decent job for these "slower" tools anyway) and just fix Sketcher to match this, I would be _**so happy.**_

obelisk79, 2022-01-20:
Hi, thanks for sharing your proposal. I've printed these out and will do some practice with them. Also, there's been discussion over on the official forums regarding sketcher keybindings you can find here: https://forum.freecadweb.org/viewtopic.php?f=3&t=41272&start=50

I'd caution to really encourage making the changes to freecad master (if possible, because you will be challenged by people resistant to change) which eventually would make it into linkstage so that interoperability between the two branches of FreeCAD won't become overly different.

It may also be possible to bundle the keybindings as a preference pack that could be easily distributed as well as a means of letting people test out this bundle of organized bindings. It would make it dead simple for people to try and provide feedback on your proposal.

brandonbeaumia, 2022-01-20:
Thank you obelisk79. I was not aware that this work was being done in FC Master, but that makes sense. Think I missed the boat already for this particular setup, but I can appreciate what Abdullah ended up landing on even if it's not my preferred style. At least it's way better than it was! Thanks for the tip about the Param bundle; I will get that together and link it from within the Spreadsheet shortly.

brandonbeaumia, 2022-01-20:
Discussion belongs in FreeCAD forum for changes to master

closed 2022-01-20 by brandonbeaumia

realthunder, 2022-02-28:
@brandonbeaumia The above commit introduces dynamic shortcut conflict resolving in FreeCAD. It can now perform longest key sequence match instead of Qt's default shortest match. In other word, Shortcut `M` and `M, M` can now coexist without any problem. I'll see if I can submit this to upstream and suggest Sketch to revert back the original shortcuts. Maybe you want to redesign your shortcut mappings without being restricted by the previous conflicts.

brandonbeaumia, 2022-02-28:
Great! I'm going to attempt building myself for all these upgrades and fixes; you've done so much work since January!

You can leave the defaults though, I'm cool with reconfiguring them if/when I need to do a fresh install. Abdullah really did put some hard thought into the new layout. It's not my style but I saw it was a group effort over on the forums and don't want to mess with that. 🙂