realthunder/FreeCAD_assembly3 #1063: Lots of rebuild issues with new daily.

HaasNL2, 2022-01-20:
Added file that is pretty broken.
If you edit Pad001 in first body, it crashes on my system.
Also lots of missing faces and sketch relations if you change anything.
[image]
[dustshoe.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7903676/dustshoe.zip)

luzpaz, 2022-01-20:
Please always attach full [About](https://wiki.freecad.org/About) info (for clarity and posterity). It's just a good habit to observe. Thanks

HaasNL2, 2022-01-21:
Sorry  
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
[dustshoe (2).zip](https://github.com/realthunder/FreeCAD_assembly3/files/7911415/dustshoe.2.zip)

Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Dutch/Netherlands (nl_NL)