realthunder/FreeCAD_assembly3 #1064: [feature request] Adding guides to lofts.

HaasNL2, 2022-01-20:
I was working with the loft feature and noticed the loft was all distorted.
[image]
It was apparently because I had 1 line to many in one sketch.
Normally I then ad some guides to it to fix this. (although removing the line was more easier, adding guides makes your model more stable mostly. (in Solidworks that was)
Example of using guides in Loft feature 
https://www.youtube.com/watch?v=s0WS__eew7E

hamza00789, 2022-01-20:
> I was working with the loft feature and noticed the loft was all distorted. [image] It was apparently because I had 1 line to many in one sketch. Normally I then ad some guides to it to fix this. [...]

I think your problem will be solved by coupling. Watch the video. 
https://youtu.be/KiccxqGMaWU?t=548 
coupling decides which point should match which point. You should also request a coupling.

hamza00789, 2022-01-20:
>                                    
Coupling benefits 
https://youtu.be/S-lATs1YtOU?t=690

HaasNL2, 2022-01-21:
@hamza00789 Looks to be the same, every CAD program I worked with had these couplings or guides.

hamza00789, 2022-01-21:
> @hamza00789 Looks to be the same, every CAD program I worked with had these couplings or guides.

Coupling and guide may be present in many cad programs. But the multi-section command in CATIA is a thousand times better than similar commands in much other software. Coupling, guide, support, etc. can also be given in it. The support in it is a very good function. This can be supported by the section as well as the guide.                  

https://youtu.be/PwUYf0B5rSE?t=201

HaasNL2, 2022-01-21:
@hamza00789 I worked 3 months in Catia v5 and I never ever wanted so hard to throw the computer out of the window. 😆
But yeah it works good.... But in the end I would give anyone who has the time or skills to add this the freedom to implement it the way he/she thinks it is best for Freecad and the community and for themself.

hamza00789, 2022-02-12:
> @hamza00789 I worked 3 months in Catia v5 and I never ever wanted so hard to throw the computer out of the window. 😆 But yeah it works good.... But in the end I would give anyone who has the time or [...]

https://www.youtube.com/watch?v=0Be-yXFvR7M  

https://www.youtube.com/watch?v=EnDx-vtJvfw  

https://www.youtube.com/watch?v=lrWyC68nrEI  

https://www.youtube.com/watch?v=2ZNF5Dsf-kc  

https://www.youtube.com/watch?v=CCRgRfHsKQg 
 
https://www.youtube.com/watch?v=g1v7lPD8Ilo 

https://www.youtube.com/watch?v=2JDVi0F9EJ0  

https://www.youtube.com/watch?v=PwUYf0B5rSE&t=221s  
   
I wanted to repeat myself, maybe all cad programs have such functions.