realthunder/FreeCAD_assembly3 #1066: Views toolbar buttons are missing with v 2022011

Crestburg, 2022-01-20:
I am a noob but installed the Windows version 2022011 when trying to follow some youtube videos when I was supposed to be using the Windows 20210717 version. I got it running ok and did the "pip" thing to fix an error message too. But when following a setup video on youtube I couldn't set the toolbar up the same because the "view buttons" are missing from the toolbar. They work fine with the 20210717 version though. Sorry if this had been reported but I signed up for github just to try to help. :)

Crestburg, 2022-01-20:
Whoops. I think it should actually be v20220111. It was missing a "1".

luzpaz, 2022-01-20:
@Crestburg in the future please add your full [About](https://wiki.freecad.org/About) info (which also tells us what OS you're using and other dependencies). Thanks

Crestburg, 2022-01-21:
Ok, here you go. Sorry about that.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

StudioPetrikas, 2022-01-21:
I think the "views" toolbar has been moved into a dropdown because we have so many redundant ways of switching views:
- View Cube (top right)
- Regular view rotation
- Hotkeys
- Pie Menus

bambukouk, 2022-01-22:
Yes, and if you really need it there is nothing stopping you customising your toolbars?

Crestburg, 2022-01-22:
Well being kind of new at this I think it would be easier to remove them, than add them. Also, the "view" is list and enabled in the toolbars list menu. But I did look at how to customize them if I really have to.
Thanks

StudioPetrikas, 2022-01-22:
> Well being kind of new at this I think it would be easier to remove them, than add them. Also, the "view" is listed as enabled in the toolbars list menu but when I disable it it removes some icons but not the ones expected. But I did look at how to customize them if I really have to. Thanks

Just to be clear - the "view buttons" are not removed, they are moved into a dropdown. By default, into "Fit selection" button.

[image]

This doesn't sound like a bug, more of a tidying up.

Crestburg, 2022-01-22:
Ahhhhh, Now I see what they did there. Ok, so just a rearrangement to consolidate buttons to unclutter the toolbar. Makes sense now. Thanks for clearing that up.

bambukouk, 2022-01-23:
Might as well close this one....

closed 2022-01-23 by luzpaz