realthunder/FreeCAD_assembly3 #1069: Body issued from a master sketch could not be mo ved (transform)

mwlandolsi, 2022-01-21:
Draw a master sketch with two rectangles for exp. using sketcher WB
create a body and pad the first rectangle
create another body and pad the second rectangle of the master sketch
the bodies could not be moved using "transform" but the position in property manager is changed.

OS: Linux Mint 20.2 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Rabbit-sk, 2022-01-24:
I can confirm that.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-24:
I can confirm that. It happens also if coordinates are changed in Data tab:

https://user-images.githubusercontent.com/74265320/150734964-aa16e160-d8dd-44a5-ae62-026be45353a0.mp4

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-24:
This issue could be renamed to **Body based on SubShapeBinder could not be moved**
[image]
[image]
[image]

hamza00789, 2022-01-24:
>  
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

I tried it works fine. I have no such problem.    
watch the video 

https://user-images.githubusercontent.com/92737149/150843708-0a8b3560-3ccc-4ee0-86b2-3eb5d13ebf69.mp4

hamza00789, 2022-01-24:
>      
And watch the video for the related post from sub shape binder move.   

https://user-images.githubusercontent.com/92737149/150844610-7822532f-68ee-477d-bc43-5e2ce783e478.mp4

hamza00789, 2022-01-24:
Looking at this issue reminded me of an old issue that was also present in the old version. #1079