realthunder/FreeCAD_assembly3 #1070: Opening a legacy document; Fixing errors; Not enough information in report view after final document recomputation

Rabbit-sk, 2022-01-22:
Hi RealThunder!

Yesterday I manged to fix all errors in an older document after opening it in Link Stable 2022.01.11. 
It was created in official branch version 0.18. 

I was dealing with: 
- swapped faces for ShapeBinders support; 
- broken geometry in sketches; 
- redundant and partly redundant constraints in sketches; 
- missing external geometry references in sketches; 
- cyclic dependencies for Pads; 
- rework of document structure from Parts and Groups to only Parts to get rid of cyclic dependency error after attempt to create external geometry in a sketch from feature inside the same body in which the sketch was (objects that were not in a praticular Group was selected in 3D view if a given Group was selected in the tree view); 

After all of that, this mesage was still displayed after opening the fixed and recomputed document:
[image]

After confirmation, a recomputation was performed immediately instead of approx. 2 minutes. 
After right-click on the document in the tree view, click on Mark to recompute, and then press F5, the recomputation was performed as usually. After saving and reopening the document, a recomputation message was displayed again. 

The solution was: 

- create a new document;
- save the document due to the next step;
- create a link to the contents of the first document in the new document; 
- imoprt all links;
- important step was done after the imoprt: recomputation of imported objects;
- delete link;
- optionally move content of Import Group to the root of the new document and delete empty Import Group;
- save the new document;

After these steps the recomputation caused no errors and no warnings in the report view.

Then I tried to open the fixed document in Link Daily 2022.01.11.
There was need to fix one SubShapeBinder and fix missing external geometry references in two sketches.

**The problem:**
After these corrections, after each recalculation of the document, there are still some warnings in the message view, but without any information where there is a problem in the model and I do not know what else to correct so that the document could be recalculated correctly. 
[image]

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

bambukouk, 2022-01-22:
I have found in the past that Link branch actually gives more and more detailed warnings than what you get in master branch
"dog leg" warnings mean that you have problem sketches (which would still be a problem in master branch but you simply wouldn't get a warning about them?)

realthunder, 2022-03-28:
I'd like to check for the repeated prompt for recomputation behavior in your description. Is if possible to reproduce the file?

Rabbit-sk, 2022-03-29:
> I'd like to check for the repeated prompt for recomputation behavior in your description. Is if possible to reproduce the file?

Thank you for looking also at older issues. I will answer your question as soon as I will have time.