realthunder/FreeCAD_assembly3 #1071: FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111 does not start

tomster, 2022-01-23:
After downloading the above mentioned release on a Windows 10 machine and executing the `RunFreeCAD` script I get the error message ```The code execution cannot proceed because MSVCP140.dll was not found. Reinstalling the program may fix this problem.``` 
I tried downloading the upstream weekly build of FreeCAD 0.20 in the hope that it might install the required DLL but while starting it worked, I still cannot start the LinkDaily version.

I even tried installing the runtime support from here (https://docs.microsoft.com/de-DE/cpp/windows/latest-supported-vc-redist) but that didn't help either.

Since I'm normally a linux user I'm now straight out of ideas. Any pointers?

bambukouk, 2022-01-23:
Did you install both x86 and x64 versions?
apparently just doing x64 doesn't work ...

tomster, 2022-01-23:
@bambukouk bingo! i had overlooked the x64 version, somehow thinking, that it pertained to the ARM platform.

To close this issue I propose to merge the PR for the README. this should avoid future issues of this nature.

luzpaz, 2022-07-11:
@realthunder thoughts on merge?

realthunder, 2022-07-13:
It's a packaging issue. The user shouldn't be required to install the runtime. The missing DLL is bundled in later release.

closed 2022-07-13 by realthunder