realthunder/FreeCAD_assembly3 #1074: The Shape color of body is not the default gray after Pad from SubShapeBinder

Rabbit-sk, 2022-01-24:
Hi RealThunder!

In 3D view the body has the same Shape color as the Shape color of SubShapeBinder. In Data  tab it has the default gray color.

[image]
[image]
FCStd and FCParam: [BodyShapeColor.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7923230/BodyShapeColor.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

hamza00789, 2022-01-24:
> 

default setting    

https://user-images.githubusercontent.com/92737149/150839739-5f5fdce0-f1ba-4fd8-9f5b-2d2f4da50d45.mp4

realthunder, 2022-03-28:
The `Color Mapping` options controls how `PartDesign` features get their colors. By default, the color is derived from the feature in history. Setting a feature's color only affects the shape that generated or modified by that feature. The `SubShapeBinder` is special, as it is not really a `PartDesign` shape feature. It has its own style setting with its view property `UseBinderStyle`. If it is `False`, then it behaves the same as other `PartDesign` feature.

With your provided file. Change the binder view property `UseBinderStyle` to false, and then manually recompute `Pad` feature will give you the gray color. If you want to have total control on the body shape color, then disable the `Map face colors` option is the way to go.