realthunder/FreeCAD_assembly3 #1081: Split Issue

hamza00789, 2022-01-24:
You may have designed it that way, but I think that's a problem. Four components are coming from the split command.  
But there must be two components.   
One solid of all the components above the surface and the other solid of all the components below the surface.   

[image]

StudioPetrikas, 2022-01-24:
I would have to disagree here. You have 2 solids. You divide each in half. You end up with 4 solids.
The split command is also great for quickly splitting unconnected bodies into parts - if you hide the cutting plane and use the `Split Tool` without it, you would end up with 2 separate bodies. I would hate losing that functionality.

hamza00789, 2022-01-24:
> 
I can't understand.  
If you look at the picture, this split command divides a pad into four parts. I want it to be two parts instead of four. Because having two parts would be more beneficial than having four parts.

hamza00789, 2022-01-24:
What are the benefits of 2 separate bodies?

StudioPetrikas, 2022-01-24:
Each solid body should have it's each separate `Body` container. Having 2 solid bodies in one `Body` container is incorrect. FreeCAD Link does not prevent you from having multiple solid bodies in one `Body`. `Split Tool` divides solid bodies.

The benefit is that it quickly allows you to 'split' the solids into separate bodies for use with `Body` containers. 

Combining the solids is trivial with Sub-Object Shape Binders.

hamza00789, 2022-01-24:
I didn't think of that, of course, it's an advantage.

hamza00789, 2022-01-25:
Shape binder takes more time. If the pattern is long. By changing the old Work Sometimes the name of the splits in the split list also changes. which causes an error. I have also worked in other software. In which split is divided into two parts and it is very good. It would be better to divide it into two parts. If it's unbearable, turn it off issue 

[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7929831/1.zip)

hamza00789, 2022-01-25:
If you have to work with a continuum without making an error, it is best to divide the split into two components.

hamza00789, 2022-01-25:
Even a waste of time can be avoided.

hamza00789, 2022-01-25:
Are you going to close it?

hamza00789, 2022-01-25:
Can you tell?

hamza00789, 2022-01-25:
Maybe I failed to explain to you any of its benefits.

hamza00789, 2022-01-25:
You may not have time to respond

hamza00789, 2022-01-25:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)               

In most cases, only two parts of the split are required.                     
As shown in the picture.   

[image]
 
In fact, in Catia I used it mostly for patterns. 
If the pattern is long then taking a shape binder one by one and boolean them can be a long task.  
If split into two parts as shown in the picture. 
And if I need a specific part, I can easily shape binder that part.

hamza00789, 2022-02-12:
https://youtu.be/8BTjPsxVwFI?t=832

hamza00789, 2022-02-12:
> Each solid body should have it's each separate `Body` container. Having 2 solid bodies in one `Body` container is incorrect. FreeCAD Link simply does not prevent you from having multiple solid bodies in one `Body`. `Split Tool` divides solid bodies.
> 
> The benefit is that it quickly allows you to 'split' the solids into separate bodies for use with `Body` containers.
> 
> Combining the solids is trivial with Sub-Object Shape Binders.

If you do not want to change the split, see it as optional. This will be very useful for the pattern. Doing more binders also makes the file larger and takes more time.                    

https://youtu.be/8BTjPsxVwFI

hamza00789, 2022-02-12:
It can also be used to cut ejector pins in mold design.

hamza00789, 2022-02-12:
> Consider me convinced :)

thanks

StudioPetrikas, 2022-02-12:
This could be the default behavior, and include a checkbox to "Split by loose parts". 
At least it's how it's called in Blender: "Separate by loose parts".

hamza00789, 2022-02-12:
> 
This is called doing parts of the pattern. Ten parts already exist in a pattern. Separating them is called split by loose parts. 

https://user-images.githubusercontent.com/92737149/153725818-b6501f56-d1cc-47db-ac47-459e84bd3cdc.mp4

Maybe I didn't understand you very well.

hamza00789, 2022-02-14:
Catia uses a reverse pocket. If you read the file I created.
I created the file with the reverse pocket in mind. 
I don't know anything about software development. 
Maybe such a technique has something to do with it. 
It may have nothing to do with [it.](https://www.youtube.com/watch?v=PoYx0B4OrLU)

hamza00789, 2022-04-09:
> 
         
I just want to know if it will work. Will there be such a feature in freecad in the future?

luzpaz, 2022-04-09:
Again, please rename this ticket a more appropriate title so we can guess what is being asked here