realthunder/FreeCAD_assembly3 #1083: visibility issue

hamza00789, 2022-01-25:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

When the body spreads, some part of the body disappears and when the body is closed, the body looks fine.   

https://user-images.githubusercontent.com/92737149/150895135-0fc98a83-d35b-4f19-b92d-556b254fe8e3.mp4

realthunder, 2022-01-25:
Can you please post the file here?

hamza00789, 2022-01-25:
> Can you please post the file here?

[Uploading 003.zip…]()

hamza00789, 2022-01-25:
Did you have this problem?

realthunder, 2022-01-25:
I think you may have installed some addon that is responsible for this behavior. Your body's `DisplayMode` is set to `Tip`, which means instead of showing the child features, the body displays its own shape that is synchronized to the `Tip` feature, i.e. the `Boolean`. However, you have also set the Boolean's `DisplayMode` to `Tools`, so if the `Boolean` feature were to be shown, it will display its tool object's shape, instead of its own shape (i.e. the shape created by the boolean operation).

I guess there is some addon implementing auto changing body `DisplayMode`, so that when collapsed, the body shows its own shape, which is the shape produced by the boolean operation. When expanded, the body shows the child features, i.e. the Boolean feature, which only shows its tool shape, because of Boolean's own `DisplayMode`.

hamza00789, 2022-01-25:
I thought it might be related to add-ons part-o-magic.          
I considered what you said but I also thought it would happen but it is not. My problem is solved in a way.   

https://user-images.githubusercontent.com/92737149/150990485-7256fe9e-4c98-4893-b79c-9a2ec02f5762.mp4

hamza00789, 2022-01-25:
Maybe it has to do with part-o-magic. it never bothered me before. It works fine when uninstalled part-o-magic. I can't live without this add-o-on. Can you tell me a solution?

hamza00789, 2022-01-25:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US) 

The problem was solved automatically for a while but I don't think it has anything to do with part-o-magic. You can see in the picture there is a split hide but the split child is visible. Note that Part O Magic is currently uninstalled.  

[image]

realthunder, 2022-01-28:
> Maybe it has to do with part-o-magic. it never bothered me before. It works fine when uninstalled part-o-magic. I can't live without this add-o-on. Can you tell me a solution?

Change `Boolean` view property `DisplayMode` to `Result`.

> The problem was solved automatically for a while but I don't think it has anything to do with part-o-magic. You can see in the picture there is a split hide but the split child is visible. Note that part-o-magic is currently uninstalled.

Change Body `DisplayMode` to `Through`. `Through` means to display the child object. `Tip` means to display the tip feature's shape, which disregards any child object's visibility.

hamza00789, 2022-01-28:
Are you talking about activating the body? I don't understand what is meant by through body. But I want to say one thing. I didn't do anything. When I opened the file today, it wasn't a problem. It seems to me that in new releases the display mode sometimes works fine and sometimes it doesn't work properly.                           

Maybe I'm misunderstanding 

[image]

hamza00789, 2022-01-28:
One thing that has come to my notice. The same problem can see in the video above. 

https://user-images.githubusercontent.com/92737149/151608438-86db6fe7-550a-4cfd-9487-39e2b2248003.mp4

realthunder, 2022-01-29:
Maybe `Part-O-Magic` has some functionality that changes the default display mode of a body. What I mean the Body's view property `DisplayMode` (better set to `Through), which you can change manually using the property view. There are also similar property `Display` in Boolean and Split (better leave it to default value, `Result`)

[image]

[image]

hamza00789, 2022-01-29:
> What is the default mode of boolean and split? (That is, of children).  

The result is the default mode of children.   

What I just understood from you is that the body's display default mode is through.

hamza00789, 2022-01-29:
What is the default mode of boolean and split? (That is, of children).  

The result is the default mode of children.   

What I just understood from you is that the body's display default mode is through.

hamza00789, 2022-01-29:
Yes, I think the result is the default mode. But as a result, it automatically goes to the tool.

hamza00789, 2022-01-29:
https://user-images.githubusercontent.com/92737149/151677096-1b4a9d2e-e5cb-41ef-a552-7c428afce228.mp4

hamza00789, 2022-01-29:
[1555.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7964706/1555.zip)

hamza00789, 2022-01-29:
In addition, this file([1555.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7964715/1555.zip)has to be updated every time it is opened without any changes, which takes a lot of time.

hamza00789, 2022-01-29:
https://user-images.githubusercontent.com/92737149/151677573-efc58c93-9b53-478d-915a-f1a306f40cb8.mp4

closed 2022-03-14 by hamza00789