realthunder/FreeCAD_assembly3 #1084: Sketcher Constrain Contextually

cuedcad, 2022-01-25:
I am not sure this it the correct place, but I am sure you have seen the new feature being offered on the FreeCAD forum, Sketcher Constrain Contextually. This looks as if it could signifcantly aid the use of Sketcher, would you consider incorporating it in one of your forthcoming releases even if
it takes time to reach master?

Thank you for the amazing developmen you are doing.

bambukouk, 2022-01-25:
It is usual and helpful to link to the discussion you are quoting:
https://forum.freecadweb.org/viewtopic.php?f=9&t=65521

cuedcad, 2022-01-25:
Apologies, I meant to

luzpaz, 2022-01-26:
PSA I feel like we should be sensitive to realthunder's workload. He is trying to merge changes upstream.

cuedcad, 2022-02-20:
Apologies for the late reply.

   Fully understand RT's workload, it is unbelievably impressive the development times!

 It was only meant to be a gentle request to keep it in mind as RT continues to
significantly improve the user's experience, but happy to wait if that is the best 
way forward.