realthunder/FreeCAD_assembly3 #1086: openning existing assembly in snap version of FreeCAD

bambukouk, 2022-01-25:
results in following long error message (BTW assembly is fine in appimage version):
```
bambuko@Z97MD3Hcd5bddbb:/snap$ free*
FreeCAD 0.20, Libs: 2022.120R14555 (Git shallow)
Adding snap-specific PYTHONPATH to sys.path: /home/bambuko/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/1/lib/python3.8/site-packages:/snap/freecad-realthunder/1/usr/lib/python3/dist-packages
connect failed: No such file or directory
<Exception> File '/run/user/1000/doc/a3927f7c/04_wheels.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/04_wheels.FCStd [File '/run/user/1000/doc/a3927f7c/04_wheels.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/14_8ft_track.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/14_8ft_track.FCStd [File '/run/user/1000/doc/a3927f7c/14_8ft_track.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/02_loco_base.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/02_loco_base.FCStd [File '/run/user/1000/doc/a3927f7c/02_loco_base.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/08_bogie_wheels.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/08_bogie_wheels.FCStd [File '/run/user/1000/doc/a3927f7c/08_bogie_wheels.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/09_tender_wheels.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/09_tender_wheels.FCStd [File '/run/user/1000/doc/a3927f7c/09_tender_wheels.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/08_bogie.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/08_bogie.FCStd [File '/run/user/1000/doc/a3927f7c/08_bogie.FCStd' does not exist!]
<Exception> File '/run/user/1000/doc/a3927f7c/07_axle_pump_hsg.FCStd' does not exist!
Exception opening file: /run/user/1000/doc/a3927f7c/07_axle_pump_hsg.FCStd [File '/run/user/1000/doc/a3927f7c/07_axle_pump_hsg.FCStd' does not exist!]
3.8e-08 <App> Document.cpp(3224): _1_loco_assembly#Link002.LinkedObject: Link not restored
Linked object: Assembly009
Linked file: 04_wheels.FCStd
0.0783552 <App> Document.cpp(3224): _1_loco_assembly#Link.LinkedObject: Link not restored
Linked object: Assembly
Linked file: 04_wheels.FCStd
0.0850565 <App> Document.cpp(3224): _1_loco_assembly#Link001.LinkedObject: Link not restored
Linked object: Assembly001
Linked file: 04_wheels.FCStd
0.0887584 <App> Document.cpp(3224): _1_loco_assembly#Link003.LinkedObject: Link not restored
Linked object: Assembly004
Linked file: 04_wheels.FCStd
0.0923961 <App> Document.cpp(3224): _1_loco_assembly#Link004.LinkedObject: Link not restored
Linked object: Assembly005
Linked file: 04_wheels.FCStd
0.0992371 <App> Document.cpp(3224): _1_loco_assembly#Link005.LinkedObject: Link not restored
Linked object: Assembly006
Linked file: 04_wheels.FCStd
0.0992947 <App> Document.cpp(3224): _1_loco_assembly#Link006.LinkedObject: Link not restored
Linked object: Assembly007
Linked file: 04_wheels.FCStd
0.099368 <App> Document.cpp(3224): _1_loco_assembly#Link007.LinkedObject: Link not restored
Linked object: Assembly008
Linked file: 04_wheels.FCStd
0.106971 <App> Document.cpp(3224): _1_loco_assembly#Link009.LinkedObject: Link not restored
Linked object: Assembly010
Linked file: 04_wheels.FCStd
0.128364 <App> Document.cpp(3224): _1_loco_assembly#Link010.LinkedObject: Link not restored
Linked object: Assembly002
Linked file: 04_wheels.FCStd
0.135073 <App> Document.cpp(3224): _1_loco_assembly#Link011.LinkedObject: Link not restored
Linked object: Assembly003
Linked file: 04_wheels.FCStd
0.159644 <App> Document.cpp(3224): _1_loco_assembly#Link012.LinkedObject: Link not restored
Linked object: Assembly
Linked file: 14_8ft_track.FCStd
0.159721 <App> Document.cpp(3224): _1_loco_assembly#Link013.LinkedObject: Link not restored
Linked object: Assembly026
Linked file: 02_loco_base.FCStd
0.159777 <App> Document.cpp(3224): _1_loco_assembly#Link014.LinkedObject: Link not restored
Linked object: Assembly027
Linked file: 02_loco_base.FCStd
0.159839 <App> Document.cpp(3224): _1_loco_assembly#Link015.LinkedObject: Link not restored
Linked object: Assembly051
Linked file: 02_loco_base.FCStd
0.159913 <App> Document.cpp(3224): _1_loco_assembly#Link017.LinkedObject: Link not restored
Linked object: Assembly034
Linked file: 02_loco_base.FCStd
0.159973 <App> Document.cpp(3224): _1_loco_assembly#Link016.LinkedObject: Link not restored
Linked object: Assembly054
Linked file: 02_loco_base.FCStd
0.160042 <App> Document.cpp(3224): _1_loco_assembly#Link018.LinkedObject: Link not restored
Linked object: Assembly081
Linked file: 02_loco_base.FCStd
0.160084 <App> Document.cpp(3224): _1_loco_assembly#Link019.LinkedObject: Link not restored
Linked object: Assembly028
Linked file: 02_loco_base.FCStd
0.160136 <App> Document.cpp(3224): _1_loco_assembly#Link020.LinkedObject: Link not restored
Linked object: Assembly033
Linked file: 02_loco_base.FCStd
0.160179 <App> Document.cpp(3224): _1_loco_assembly#Link021.LinkedObject: Link not restored
Linked object: Assembly083
Linked file: 02_loco_base.FCStd
0.178668 <App> Document.cpp(3224): _1_loco_assembly#Link031.LinkedObject: Link not restored
Linked object: Assembly001
Linked file: 08_bogie.FCStd
0.18316 <App> Document.cpp(3224): _1_loco_assembly#Link032.LinkedObject: Link not restored
Linked object: Assembly075
Linked file: 07_axle_pump_hsg.FCStd
0.183247 <App> Document.cpp(3224): _1_loco_assembly#Link033.LinkedObject: Link not restored
Linked object: Assembly077
Linked file: 07_axle_pump_hsg.FCStd
0.183311 <App> Document.cpp(3224): _1_loco_assembly#Link034.LinkedObject: Link not restored
Linked object: Assembly078
Linked file: 07_axle_pump_hsg.FCStd
0.183386 <App> Document.cpp(3224): _1_loco_assembly#Link035.LinkedObject: Link not restored
Linked object: Assembly079
Linked file: 07_axle_pump_hsg.FCStd
0.185222 <App> Document.cpp(3224): _1_loco_assembly#Link023.LinkedObject: Link not restored
Linked object: Assembly038
Linked file: 08_bogie_wheels.FCStd
0.25807 <App> Document.cpp(3224): _1_loco_assembly#Link024.LinkedObject: Link not restored
Linked object: Assembly039
Linked file: 08_bogie_wheels.FCStd
0.342313 <App> Document.cpp(3224): _1_loco_assembly#Link029.LinkedObject: Link not restored
Linked object: Assembly002
Linked file: 09_tender_wheels.FCStd
0.373547 <App> Document.cpp(3224): _1_loco_assembly#Link026.LinkedObject: Link not restored
Linked object: Assembly
Linked file: 09_tender_wheels.FCStd
0.405063 <App> Document.cpp(3224): _1_loco_assembly#Link028.LinkedObject: Link not restored
Linked object: Assembly001
Linked file: 09_tender_wheels.FCStd

```

ppd, 2022-01-25:
This issue has been resolved in the forum. It's due to the document portal only giving access to the file the user selected for security reasons. For files in the home directory, this does not apply, and so linked files can be found just fine.

bambukouk, 2022-01-25:
Well... not so quickly 😉
I have copied the whole project folder into /home/documents folder, to see if this avoids issues - and, no, the result was the same.

Plus, I have no problems openning files from my NAS mounted folder as long as they are not externally linked, so the location etc is not an issue.

The problem is openning externally linked files in assembly.

So, no, the problem is not resolved

ppd, 2022-01-25:
Is your home interface connected? The problem is that selecting a file only gives access to that file in the flatpak/snap security model unless you have access to that directory anyway.

bambukouk, 2022-01-25:
Are you talking about some access setting in snap?
I simply do not uderstand you - sorry 
I have a bunch of files in a folder - all part of externally linked assembly.
If I can open one file in this folder why cannot it open others?

ppd, 2022-01-25:
I understand you perfectly fine and our understanding of external links is the same. I responded in the forum.

bambukouk, 2022-01-25:
Sounds like I will have to do more reading....
or just give up and continue using appimage - at least it is working fine 👍

ppd, 2022-01-25:
If it works, it works :). Maybe it's an issue with an older portal version on your Ubuntu version. That's one of the few things that are not updated regularly in the snap ecosystem.

bambukouk, 2022-01-25:
I had to install snap before using it on my EOS 6, so it should be up to date?

ppd, 2022-01-25:
That's 20.04 based, so not too bad. I think I'll give it a try in a VM tomorrow and check if I can reproduce the issue.

bambukouk, 2022-01-25:
no rush 😀
and thank you for your support!

bambukouk, 2022-01-25:
BTW there is one more thing that might be connected to my problem
The association between FreeCAD and .FCStd files is missing.... and I seem unable to set it?
So, I can open .FCStd files from within FreeCAD (is that your "portal" thingy?) but when I try to open them outside, my Linux doesn't know what to do with them...??

ppd, 2022-01-26:
Tests in VMs paint the following picture:
The behaviour in older portal versions, as present on Ubuntu 20.04, is indeed different. It identifies the cifs mount as a folder that the snap has no access to, and therefore it proxies the access over the document portal. In this case, FreeCAD sees translated paths like `/run/user/1000/doc/assembly.fcstd`. 

On more recent Ubuntu versions, this is not the case. 
I'm wondering whether some mount option can fix this; I'll ask on the snapcraft forums.

To put it in a nutshell: The file chooser should return something like `/home/user/mountdir/assembly.fcstd` to FreeCAD. It believes `/home/user/mountdir` is not accessible by the snap, so it proxies the `assembly.fcstd` to the snap because the user explicitly chose to open _this_ file. The snap gets it proxied as `/run/user/1000/doc/assembly.fcstd`, so it searches for the links as `/run/user/1000/doc/link.fcstd` etc., which obviously fails. If you opened the assembly from the cmd line like `freecad-realthunder ~/remotedir/assembly.fcstd` everything would work because the file chooser is not involved.

> The association between FreeCAD and .FCStd files is missing.... and I seem unable to set it?

That is a limitation of snaps at the moment. They can't register new MIME types. See https://forum.snapcraft.io/t/allow-snaps-to-register-new-mime-types/6467

bambukouk, 2022-01-26:
Wow!
Very comprehensive answers - thank you
I will try assembly from terminal to confirm 
Sounds like my problem is my choice of distro 👎

ppd, 2022-01-26:
Just a very unfortunate corner case. The use-case for the file portal is more like "open exactly one file. This is what the user wants", whereas FreeCAD is thinking in terms of "point me to a file. I will find its dependencies relative to it.". So the older desktop portal thinks it's doing the right thing, and FreeCAD also.

Of course, this still works for the majority of the users because they just work on local files, where both use-cases are functionally the same.

bambukouk, 2022-01-26:
Well... it was an interesting exercise and I guess it helped to discover some limitations of snap.

ppd, 2022-01-26:
One crude solution would be to force Qt to use its native file dialog. This would solve the problem, too. Maybe the platformtheme can be overridden.

bambukouk, 2022-01-26:
My thinking by now is : what advantages does snap offer, so that I would want to persevere with increasingly complex and convoluted attempts to make it work....
For me it may simply be the case of carry on with appimages? 
If it ain't broke don't bother trying to fix it 😀

ppd, 2022-01-26:
Ah, you can force it:
[image]

Then it'll behave like the AppImages do.

ppd, 2022-01-26:
That's a good candidate for an FAQ.

bambukouk, 2022-01-26:
Yes, it works :+1:  indeed.
Nice find.

ppd, 2022-01-26:
@realthunder so this is a problem if the portal is too old and the user selects a file containing links. There's the possibility of using the internal file dialog by default (see the build option). What do you think is the best option as the default?

realthunder, 2022-01-26:
There is a build option USE_QT_FILEDIALOG, but still can be overridden by user parameter (shown in your screenshoot), and the those parameters maybe shared by other FreeCAD installation, and therefore cannot be relied up on. Are there anyway to identify the portal version at runtime? If so, I can add code to force use qt file dialog if necessary.

ppd, 2022-01-26:
Possbily via dbus. I'll have to look into it. 

Another problem I'm just seeing now with older portals: The portal only gives access to the exact filename the user chose when saving, e.g. they type "my_assembly". FreeCAD automatically appends the FCStd suffix _after_ getting this filename back from the portal. This is not allowed because the portal only mounts "my_assembly" into the snap/flatpak. my_assembly.FCStd is treated as an unrelated, temporary file, and it does not propagate into the "outside world".

This is a bigger issue on Gtk desktops as the Gtk portal does not respect the default file suffix that Qt sends. Or maybe Qt does the wrong thing, I don't know for sure.

ppd, 2022-01-26:
It seems it's only possible to query the interface version over dbus, but not the implementation version, which would be the crucial information. The only bulletproof method would be to reject file paths that begin with `/run/user/$UID` in the file dialogs and give the user the option of selecting via the internal dialog :man_shrugging:

ppd, 2022-01-28:
Related:

- Bug filed for default filename support in Qt's QXdgDesktopPortalFileDialog: https://bugreports.qt.io/browse/QTBUG-100297
- https://github.com/ppd/freecad-ppd/commit/11afa4bbe86546dfb806e2afcbba0a2ddea4d446

bambukouk, 2022-01-29:
I guess we can close this one? 
with all the actions taken by @realthunder  and @ppd  ?

closed 2022-01-29 by bambukouk