realthunder/FreeCAD_assembly3 #1087: Toggling visibility issue

DerTFL, 2022-01-25:
OS: KDE neon User - Plasma 25th Anniversary Edition (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Russian/Russia (ru_RU)

OS: KDE neon User - Plasma 25th Anniversary Edition (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Russian/Russia (ru_RU)

I have tested both version and have same behavior. After save and open model i can't hide some object pad's, sketch's, etc...

On release 2021.1015.24301 i don't have this problems.

[test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7936847/test.zip)

realthunder, 2022-01-25:
'CTRL + click' the green eye icon to toggle the 'Show on top' of those objects. You can hover your mouse to the eye icon for more information.

DerTFL, 2022-01-26:
not 'CTRL + click' or 'ALT + click' not working for this objects.

m-sundman, 2022-01-26:
I can confirm this. When creating a new project and new objects will function just fine, but after saving and reloading the project it's no longer possible to hide objects. There's also some problem changing the draw style.

Here I've hidden literally everything, and the mouse pointer isn't over anything, and I'm not touching the keyboard, but still everything is visible:
[image]

m-sundman, 2022-01-26:
I think I've found a workaround: After loading a project you can Alt+click on all green eye icons and then objects can be hidden again. (Until the next time you save and load.)

realthunder, 2022-01-27:
The problem should be fixed by the above commit. Feel free to re-open if otherwise. A temporary workaround is to press `Ctrl+Shift+Space` while there is no selection or pre-selection. This shall clear any show on top object.

BTW, can you please tell me how you get those show on top object saved to file, because apparently you didn't do it intentionally? Do you have a habit of saving project while editing some object?

closed 2022-01-27 by realthunder

m-sundman, 2022-01-27:
> can you please tell me how you get those show on top object saved to file, because apparently you didn't do it intentionally?

I had no edits open when I saved. I just created a new project, added a body with a sketch that I revolved, added another body with a sketch I extruded, and after that I saved it. I didn't turn any of those eyes green in any way. I didn't press any keyboard shortcuts and didn't select anything in the model tree. I didn't have any edits open, although the second body might have been active. But the problem was with both bodies, and 2 bodies can't even be active at the same time, so...