realthunder/FreeCAD_assembly3 #1089: (not an issue - question) how do I (use images in externally linked assemblies)

bambukouk, 2022-01-26:
Most (if not all) of my designs use (correctly scaled and inserted in correct location and plane) `images` as a basis of design
When all my assemblies were done in a single file, it was simple - all images were in one file and I could use them for any sub-assembly I needed.

I have since moved to externally linked assemblies - works great and I am very happy using it but...
I have a problem with the `best way of using images across now different/separate files for sub-assemblies`.

In theory I could insert (repeatedly) all the images I need in all the files they are needed, but it would be a right PITA scaling and transforming them every single time...

Is there a better way? (I can't `link` them?)
is there a way of using an image with the same scale and placement in more than one file?
am I missing something?
Help, please.

realthunder, 2022-01-27:
I am not sure I get what you want here. Can you please make a sample file showing your intention, like a single file assembly where an image is used in multiple places?

Maybe the `LinkTransform` property of the link fits your need? By default, a Link override the linked object's placement, so that you can get independent placement. If you set `LinkTransform` to `True`, then the link will track the linked object's placement.

In case the image is inside a container (e.g. assembly container), and you want to follow the image's placement in the global cooridnate system (i.e. the accumulated transformation from the top parent to the image), you can use a `Sub-Link`.

[image]

bambukouk, 2022-01-27:
I am sorry for messy description.
I will try your suggestions (thank you :+1: )
and I will prepare sample file

bambukouk, 2022-01-27:
Solved - sorry to have bothered you with such trivia.
Simply placing image inside assembly container (which never occurred to me...) solves my problem.
I can now simply link images used in sub-assembly files to main assembly file with all the usual advantages.
The only thing is that you cannot freeze assembly containing the image, but this is not really any problem - just a case of organising/separating assembly containers in a sensible manner.
Thank you for all you doing for all of us!
[image_test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7950216/image_test.zip)
!

closed 2022-01-27 by bambukouk