realthunder/FreeCAD_assembly3 #1092: The attached step file loads but shows no colour, works with FreeCAD 0.19 & 0.20

KeithSloan, 2022-01-28:
The attached step file loads but shows no colour, works with FreeCAD 0.19 & 0.20
[dm1-id-214.stp.txt](https://github.com/realthunder/FreeCAD_assembly3/files/7962299/dm1-id-214.stp.txt)

hamza00789, 2022-01-30:
https://user-images.githubusercontent.com/92737149/151709716-3e51d317-f836-47c1-a3cd-240c7e6a6cc3.mp4      

file colors work fine

KeithSloan, 2022-01-30:
Not here they don't

OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: C/Default (C)
[image]

luzpaz, 2022-01-31:
@hamza00789 please X 3 
**Always** add your full [about](https://wiki.freecad.org/About) info. It saves extra confusion and troubleshooting when you do.

kevenwyld, 2022-02-02:
I think you need to update, you are on the 10.15 version and while there isn't a specific item in the changelog I believe this was resolved in the next release here:  https://github.com/realthunder/FreeCAD_assembly3/releases

There were several previous issue about color representation in the last few releases that don't appear to exist in the latest.

It works for me on:
```
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

luzpaz, 2022-04-14:
Initially I could reproduce the 'bug' but then with the help of @studiopetrikas I understood it was STEP preferences. Once i adjusted them I saw the colors.
@KeithSloan What are you STEP settings? 

Here are mine
[image]
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

KeithSloan, 2022-04-24:
Sorry was on holiday which has turned into COVID self isolation - Will look at when feeling better.

luzpaz, 2022-04-25:
Feel better. Let us know how you're doing!

goeland86, 2022-05-04:
Hi, just want to add to this - I got a .stp file from grabcad to import and work with. When I try to open or import the stp file, the FreeCAD stable and daily builds from April crash.

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.406.28575 (Git)
Build type: Release
Branch: LinkStage3
[X MOTOR GROUP.stp.txt](https://github.com/realthunder/FreeCAD_assembly3/files/8618709/X.MOTOR.GROUP.stp.txt)

Hash: 0ac46d957b0b98af48c7a94dde51e518d1388f31
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```
[X MOTOR GROUP.stp.txt](https://github.com/realthunder/FreeCAD_assembly3/files/8618710/X.MOTOR.GROUP.stp.txt)

The stp file generating the crash can be found here: https://grabcad.com/library/step-motor-nema-17-1 (tried to attach it as .txt but it seems to be too large?)

realthunder, 2022-05-04:
Please try again with the latest release [here](https://github.com/realthunder/FreeCAD/releases).

goeland86, 2022-05-04:
@realthunder Thank you, I hadn't noticed there was a new release. It loads fine now, thank you!