realthunder/FreeCAD_assembly3 #1093: FreeCAD_assembly3 with python 3.10 crashes

dravac, 2022-01-29:
I compiled Freecad with python 3.10  
The compilation ends successfully.  

Frecad is launched correctly except FreeCAD_assembly3 does not work:  

```
During initialization the error "PY_SSIZE_T_CLEAN macro must be defined for '#' formats" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 158, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/maison/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/init_gui.py", line 3, in <module>
    from . import gui
  File "/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/maison/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/gui.py", line 302, in <module>
    class AsmCmdNew(AsmCmdBase):
  File "/home/maison/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/proxy.py", line 206, in __init__
    mcs.register(cls)
  File "/home/maison/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/gui.py", line 213, in register
    FreeCADGui.addCommand(cls.getName(),cls)
Base.FreeCADError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats
--------------------------------------------------------------------------------
```

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.30792 (Git)
Build type: Release
Branch: LinkDaily
Hash: ec0fcee41ff6847a7649fef18b0dc00202a52a27
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)

bambukouk, 2022-01-29:
similar?
https://forum.freecadweb.org/viewtopic.php?f=4&t=65608

dravac, 2022-01-29:
I think more like this:  
 The PY_SSIZE_T_CLEAN macro must now be defined
[https://docs.python.org/3.10/whatsnew/3.10.html#id2](url)
[https://docs.python.org/3.10/c-api/arg.html#strings-and-buffers](url)
Add `#define PY_SSIZE_T_CLEAN` before` #include <Python.h>`
But what files are involved ?

dravac, 2022-02-06:
- coin3D/pivy
[https://github.com/coin3d/pivy/commit/2f049c19200ab4a3a1e4740268450496c12359f9](url)
I compiled pivy with the update then freecad
but there is always the same error PY_SSIZE_T_CLEAN.
Frecad is launched correctly except FreeCAD_assembly3 does not work

realthunder, 2022-02-06:
I have updated the repo to fix for Python 3.10. I use travis for auto build and uploading to pypi. But I Just found out that my travis credits runs out. I have already sent request to ask for more credits. Wait till it gets updated, or maybe you want to try to [build](https://github.com/realthunder/slvs_py/issues/1#issuecomment-720028439) it yourself.

luzpaz, 2022-02-07:
Relevant: https://github.com/FreeCAD/FreeCAD/pull/5503

luzpaz, 2022-02-07:
@realthunder maybe we can ask @berndhahnebach if we can use FreeCAD CI for Link builds ?

realthunder, 2022-02-07:
You mean for Link branch release? I don't know, that may be too much to ask. Or, you mean the `py-slvs` module involved here? That's a small build, I think I can even setup using github's CI for it. It's just that I have already setup for travis to do the job, and don't want to waste time converting it for the moment.

luzpaz, 2022-02-07:
Yea, for Link branch.

dravac, 2022-02-07:
- py-slvs
https://github.com/realthunder/solvespace/commit/c21cb07b89057f3785529ba0dab0cf00160aa687
I've compiled following the method:
https://github.com/realthunder/FreeCAD_assembly3/issues/279#issuecomment-765859659
but there is always the same error PY_SSIZE_T_CLEAN

Piezoid, 2022-02-08:
> Relevant: [FreeCAD/FreeCAD#5503](https://github.com/FreeCAD/FreeCAD/pull/5503)

In my tests, this PR solved the OP's issue for both mainline and LinkDaily.
This [line, in Gui module](https://github.com/FreeCAD/FreeCAD/blob/cafb8601e003ca3def860142fad430c824b5a191/src/Gui/ApplicationPy.cpp#L1223) is where the crash is triggered.

realthunder, 2022-02-13:
@Piezoid I have cherry picked fix for this in https://github.com/realthunder/FreeCAD/commit/91c5e00f6fd7c994bb7e73c188f6bad57a4928eb

@dravac There is update for py-slvs on pypi. Can you please update it with pip and try again? If you are using AppImage, you can update it like this

```
./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage pip install --upgrade --user py-slvs
```

dravac, 2022-02-15:
Thanks 
I compiled freecad and it works fine.  
The error PY_SSIZE_T_CLEAN is fixed.
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.30816 (Git)
Build type: Release
Branch: LinkDaily
Hash: 11e5cb0cd61ef5abbae31d5ce424299fb9c7a154
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)
```

closed 2022-02-15 by dravac