realthunder/FreeCAD_assembly3 #1096: Cannot add point as element

not-a-number-dot-org, 2022-01-30:
New file
Save
Part Design -> Create body
(sometimes) generates warning :
<PartDesign> CommandBody.cpp(150): Ignore selection with no solid shape: Sans_nom1#Body.

Create sketch in body and pad it (cube for example)
Create assembly, drag body to parts,
select a point (cube corner for example)
Drag it to Elements (an Element is created)
generates errors :
<Part> assembly.py(896)|ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of Sans_nom#Element
<Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of Sans_nom#Element

Select the element created.... corner not highlithed in 3d view

Also... in a complex assembly (many parts), all the PointCoincident Constaints are displayed at the same place in the 3d view and not on the parts they belong to. My assemblies with PointCoincident constraints does not resolve anymore.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

Rabbit-sk, 2022-01-30:
Maybe related: #1049 #1076

realthunder, 2022-03-28:
Confirm same as #1129. Should be fixed now.

closed 2022-03-28 by realthunder