realthunder/FreeCAD_assembly3 #1097: Integrate an assembly containing subassemblies into a higher level assembly

jpg87Fr, 2022-01-31:
Hello Realthunder,
I went several months without really using FreeCad, including your branch.
I have just taken up a question which had been asked on the forum and to which I had stupidly answered that this possibility would be a luxury!
https://forum.freecadweb.org/viewtopic.php?p=448913#p448913

As you answered that you had an idea, were you able to develop it? Have you added a property that allows you to integrate into a global assembly not only sub-assemblies as kinematic equivalence classes (a single block), but also sub-assemblies themselves containing kinematic equivalence classes (therefore mobile between them) without blocking everything?

As an example, I propose the Theo Jansen Sand Walker, which I believe is the mechanism offered in your wiki as an example skeleton sketch.

In this post,
https://forum.freecadweb.org/viewtopic.php?p=536291#p536291

Zolko makes this system work in Assembly 4, and it would be great to find a way to make it work in Assembly 3 too
[Pieces_TheoJansen_LinkBranch.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7973416/Pieces_TheoJansen_LinkBranch.zip)
.

bambukouk, 2022-02-01:
Easy to make `one` work:
[image]

File showing the way I did it (different to you @jpg87Fr  :+1: ) attached below:
[legs(bambuko)07.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7980052/legs.bambuko.07.zip)

but I agree with all of you - how do you do top level assembly using links?

jpg87Fr, 2022-02-02:
There is no problem since everything happens in the same assembly: all your kinematic equivalence classes are at the same level.

bambukouk, 2022-02-02:
Yes, I know...
Let's hope that @realthunder  can magic some way of overcoming this limitation?