realthunder/FreeCAD_assembly3 #1098: The Assembly3 Workbench isn't listed in FreeCAD Workench list

icaro-meneses, 2022-02-02:
I can't find the Assembly3 workbench on the list. My FreeCAD installation was done through the official Arch Linux repository. 
I did manually install the Assembly3 package via git clone in the folder that was listed in the installation instruction. Also I installed `SolveSpace` Python module. Although I did as the instructions said, I did not get the correct result and also do not know how to proceed in this case.

It said: `/home/<your_user_name>/.FreeCAD/Mod`

[image]

My system information:
```
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: makepkg
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

realthunder, 2022-02-02:
It looks like AUR build does not pre-bundle Assembly3 workbench, which is expected. 

> Also I installed `SolveSpace` Python module. Although I did as the instructions said, I did not get the correct result and also do not know how to proceed in this case.

Are there any error message? When you restart FreeCAD, are there any warning message?

icaro-meneses, 2022-02-02:
> It looks like AUR build does not pre-bundle Assembly3 workbench, which is expected.

But I installed it through the official Arch Linux repository, not the Arch Linux User Repository (AUR).

> Are there any error message? When you restart FreeCAD, are there any warning message?

There was no error message or warning message. When I restart FreeCAD, everything seems normal until I go on the list of workbenches and realize that Assembly3 is not there.

icaro-meneses, 2022-02-02:
I was curious to start FreeCAD through the terminal, and I got this message:

```
Assembly4 workbench (v0.11.7) loaded

(freecad:1694): Gtk-WARNING **: 07:02:12.329: Unable to locate theme engine in module_path: "adwaita",

(freecad:1694): Gtk-WARNING **: 07:02:12.329: Unable to locate theme engine in module_path: "adwaita",

(freecad:1694): Gtk-WARNING **: 07:02:12.331: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.331: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.331: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.332: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.332: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.332: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.332: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.333: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.333: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.333: Unable to locate theme engine in module_path: "murrine",

(freecad:1694): Gtk-WARNING **: 07:02:12.333: Unable to locate theme engine in module_path: "murrine",
During initialization the error "PY_SSIZE_T_CLEAN macro must be defined for '#' formats" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/icaro/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/init_gui.py", line 3, in <module>
    from . import gui
  File "/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/icaro/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/gui.py", line 302, in <module>
    class AsmCmdNew(AsmCmdBase):
  File "/home/icaro/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/proxy.py", line 206, in __init__
    mcs.register(cls)
  File "/home/icaro/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/gui.py", line 213, in register
    FreeCADGui.addCommand(cls.getName(),cls)
Base.FreeCADError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats
--------------------------------------------------------------------------------
connect failed: No such file or directory
```

In this message, something indicates that there is some problem with some files contained in the Assembly3 module directory. I hope this helps us solve the problem.

luzpaz, 2022-04-11:
Please retest with upstream's latest [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)  
Assembly3 is now available through the Addon Manager. 

Closing ticket. If issues persist please re-open

closed 2022-04-11 by luzpaz