realthunder/FreeCAD_assembly3 #1099: FCStd opens correctly on Windows but not Linux (same version number)

brandonbeaumia, 2022-02-02:
Offending 13MB file (rename to MDT.fcstd): [MDT.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7983443/MDT.zip)

This file was created on Windows, and the 2 bevel gear STEP files were imported while in that environment. It seems like it's continuing to work correctly on Windows: everything shows up, no broken geometry. On Linux, the STEP files that were imported into this document do not show up. They are listed in the tree view, but have no representation in the 3D viewport and the datum planes that were attached to some of the sketch geometry error out with a variation of `PositionBySupport: AttachEngine3D: subshape not found BaseFeature.Face9`.

Linux about:
OS: Pop!_OS 21.10 (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Windows about:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

realthunder, 2022-02-02:
The problem is fixed by the above commit. When the next Windows build is available, you can fix your file by open it in Windows and save again, and Linux version will be able to show the content.

closed 2022-02-02 by realthunder

brandonbeaumia, 2022-02-02:
Wow, that was fast; thank you!