realthunder/FreeCAD_assembly3 #1103: Bug: PlaneCoincident Lock Angle random behavior

not-a-number-dot-org, 2022-02-02:
Hello.

I have a very strange behavior with Lock Angle of PlaneCoincident constraint.
It seems to work but suddenly (meaning with no reason it stops working.
I discovered this while asking for help on Assembly forum of FreeCAD.
So everything (including file) is already explained there : https://forum.freecadweb.org/viewtopic.php?f=20&t=65900&p=567699#p567699

Thanks

luzpaz, 2022-07-01:
@not-a-number-dot-org is this still an issue on latest Daily ?

not-a-number-dot-org, 2022-07-02:
Yes it looks like.
Here is a sub example with only the nut block and the screw if you're willing to test.
Sorry I wont have time to fully test it until end of July.

[test.FCStd.gz](https://github.com/realthunder/FreeCAD_assembly3/files/9033275/test.FCStd.gz)

not-a-number-dot-org, 2022-08-06:
Just installed 2022.709

from angle 0° to 80° with 10° stepping : it works (<<vis sans fin>>.Placement.Angle is 80°)
back from 80° to 0° in one step : doesn't work (<<vis sans fin>>.Placement.Angle is set to 90° instead of 0°)

realthunder, 2022-08-07:
With current implementation of the SolveSpace solver, this is an expected behavior. To work around angle ambiguity problem, use `Attachment` constraint whenever possible. You can change the angle by changing the `Offset` property of any of the `ElementLink` grouped under the `Attachment` constraint.