realthunder/FreeCAD_assembly3 #1105: can not to create sketch directly on face and unable to use external geometry for some parts

smarq8, 2022-02-03:
As I try to create sketch on some face then it is placed by some offset and I need to reattach it manually in 'attachment editor'.
I also can not succesfully to use external geometry on some objects, as you can see it sometimes work or not.

These two boards are imported do FC by kicadstepup plugin and positioned by assembly3

[image]
[IMU cube.zip](https://github.com/realthunder/FreeCAD/files/7993799/IMU.cube.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Polish/Poland (pl_PL)

bambukouk, 2022-02-03:
When I try to open `IMU cube.FCStd` it doesn't work for some reason?
It opens but lots of error messages (in red):
```
14:24:28  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound
14:24:28  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound001
14:24:28  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound002
14:24:28  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound003
14:24:28  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound004
```
It may not be you and your file - this is the second one I am having problem with.
Can someone else, please try opening it?

smarq8, 2022-02-03:
for me its openable only for FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111
I laso got similar error at FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111

brandonbeaumia, 2022-02-04:
Same errors here.

`22:33:18  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound
22:33:18  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound001
22:33:18  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound002
22:33:18  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound003
22:33:18  <Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of IMU_cube#Compound004`

smarq8, 2022-02-04:
very strange
I already try to clear unzip FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111 and clear any config data at %appdata%/FreeCAD but still I can open it with only that version
The only difference I can see is different OS,python version and OCC version however FC claim to use "Python version: 3.8.6+" while my OS say something different "Python 3.9.7 (default, Sep 23 2021, 09:13:41)  [GCC 10.3.0 64 bit (AMD64)] on win32"

closed 2022-02-04 by realthunder

realthunder, 2022-02-04:
@smarq8 The problem described in your OP only occurs for shape without topo naming history, e.g. shape imported from STEP, and the referenced commit above fixed problem.

As for not being able to see the shape with other version of FreeCAD is because you author the file with the latest Windows daily release, which uses OCC 7.6, while all the other release uses OCC 7.5. The problem has been reported in #1099, and has been fixed already.