realthunder/FreeCAD_assembly3 #1107: Cannot open Stls after they are created or other stls

Sturmrider, 2022-02-03:
I have the latest version of the Realthunder FreeCad. Everytime I attempt to open an STL file whether I created it or it comes from another maker, Freecad just sits for about a minute or two and then just disappears. I will attach a file I got from Prusa and one I created. I am working on Windows 10 Dell XPS with 16gb of Ram. 

I did not have this issue with the previous version of the Realthunder link so I'm not sure what is different. I get no error message so it is a mystery to me.

Thanks for any help you can give.

Regards,

Larry Whalen

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)
[RftG_Card_Shoe_Sleeved.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7995305/RftG_Card_Shoe_Sleeved.zip)
[Card Storage-Bottom HalfFillet001.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7995308/Card.Storage-Bottom.HalfFillet001.zip)

bambukouk, 2022-02-03:
Both open fine in a matter of seconds...

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```
[image]

Sturmrider, 2022-02-03:
Thank you for your quick reply. Just tried opening again and failed when I reopened FreeCad I had this error message:

10:02:31  2.675583 <asm3.main> init_gui.py(14): no solver backend found

Not sure what this means. Thank you again for your help.

Regards,

Larry

bambukouk, 2022-02-03:
You need to install the solver.
Should have happened when you were installing the latest release of FreeCAD

Sturmrider, 2022-02-03:
Where do I find the solver and how do I install it. I read the instructions on the page, but I never got a prompt to install it. It says the py-slvs is to be installed using pip. I am not familiar with pip and do not know where to find the py-slvs.

Thank you again for your help. This one I am a little lost on. I have been doing okay otherwise with FreeCad.

Regards

Larry

bambukouk, 2022-02-03:
Have you used assembly3 before?
or is it the first time you trying both FreeCAD and assembly3?

Sturmrider, 2022-02-03:
I have used FreeCad quite a bit and I am seeing the Real Thunder Version is in a directory called FreeCAD-asm3, but I have not used assembly 3 before I am guessing. I have followed the instructions on the web site below is what I am getting for a message, but still no success opening STLs I am missing something. I do not have the directory c:\Users\Larry\AppData\Roaming\FreeCAD\Mod on the computer and I have version 19 installed as well as the Real Thunder version. I have been using 19 to open STLs I would prefer to go back to just one version on the computer. 

Thank you again for your help.

Microsoft Windows [Version 10.0.19042.1466]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Larry>cd C:\Users\Larry\Desktop\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\bin

C:\Users\Larry\Desktop\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\bin> python.exe -m pip install py_slvs
Collecting py_slvs
  Downloading py_slvs-1.0.2-cp38-cp38-win_amd64.whl (263 kB)
     |████████████████████████████████| 263 kB 1.1 MB/s
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.2
WARNING: You are using pip version 20.3; however, version 22.0.3 is available.
You should consider upgrading via the 'C:\Users\Larry\Desktop\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\bin\python.exe -m pip install --upgrade pip' command.

bambukouk, 2022-02-03:
this is getting well beyond my pay grade :-)
not to mention the fact that I am not using Win (I am running Linux)

I wasn't asking whether you were using FreeCAD a lot :-) just trying to establish whether you had Assembly 3 installed in some previous version of realthunder's Link branch? 

I guess there would be nothing wrong with re-installing the latest release of realthunder's Link branch again....

Sturmrider, 2022-02-03:
I'll give that a try. Thanks for all your help you went above and beyond.

realthunder, 2022-02-04:
Can you please try [this](https://drive.google.com/file/d/1aVGnKlVCGe7Sr6ZxOMMNdRo58bmiEgPv/view?usp=sharing) build, and let me know the result?

realthunder, 2022-02-04:
> Is there anywhere I can contribute something to support your efforts as well.

Maybe you can become one of my patrons in [Patreon](https://www.patreon.com/thundereal)?

As for the addon error, please start FreeCAD using the `RunFreeCAD.bat` file in the root of the extracted folder.

Sturmrider, 2022-02-05:
You will also find you have a new Patron.:)

herrgerd, 2022-02-08:
> > Is there anywhere I can contribute something to support your efforts as well.
> 
> Maybe you can become one of my patrons in [Patreon](https://www.patreon.com/thundereal)?
> 
> As for the addon error, please start FreeCAD using the `RunFreeCAD.bat` file in the root of the extracted folder.

Btw @realthunder what service does take the least ammount off of our donations? Is there any better (cheaper for us, more money for you) service?

realthunder, 2022-02-08:
I haven't done any comparison yet. I guess it also depends on which contry the donors come from. I heard [liberapay](https://liberapay.com/realthunder/donate) is better in some sense.

luzpaz, 2022-04-15:
Original issue solved. JFYI, A3 is available via Addon Manager now an independent of Link branch.

closed 2022-04-15 by luzpaz