realthunder/FreeCAD_assembly3 #1110: Non-stable "inconsistent constraints" issues with the geometry.

goeland86, 2022-02-07:
As requested on the forum thread I'm opening the issue here. (Thread with screenshots: https://forum.freecadweb.org/viewtopic.php?f=20&t=66061)

What's happening is when I try to constrain the MGN carriage (from a step, but saved as a FreeCAD file, and linked into the assembly) with the delta rod & belt carriage part (modeled in FreeCAD and linked in the assembly), I manage it on tower 1.  See screenshot:
[image]
Then it throws the error
```
20:28:20  0.001001 <asm3.sys> system.py(244): auto relax constraint (MonsterKossel_assembly#AxialAlignment024) MonsterKossel_assembly#Constraint101 between (MonsterKossel_assembly#Carriage2) MonsterKossel_assembly#Link035 and (MonsterKossel_assembly#MGN9CZFC2) MonsterKossel_assembly#Link033, 2
20:28:20  9.545851 <asm3.main> assembly.py(3918): solver exception when auto recompute
Traceback (most recent call last):
  File "C:\Users\goela\Desktop\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\Mod\asm3\freecad\asm3\solver.py", line 110, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "C:\Users\goela\Desktop\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111\Mod\asm3\freecad\asm3\sys_slvs.py", line 70, in solve
    raise RuntimeError(reason)
RuntimeError: inconsistent constraints
```
for tower 2 or 3, even though I'm trying to set up the same constraints, as shown in this screenshot:
[image]

I have a wild guess that because I'm using links to the original FreeCAD file and not to copies of the objects in the main assembly file itself, it's causing issues if I re-use the same points of the geometry for constraint setup?

All the files to replicate and investigate are on the github repository (latest master): https://github.com/goeland86/MonsterKossel/
Until this issue is understood, I won't push anything to the repo.

I suspect user error rather than software issues, so I look forward to your input on the matter.

luzpaz, 2022-02-07:
Please edit the OP and add your full [About](https://wiki.freecad.org/About) info. thx!

goeland86, 2022-02-07:
@luzpaz done, sorry for missing out on that.