realthunder/FreeCAD_assembly3 #1111: <asm3.main> init_gui.py(14): no solver backend found

natrius, 2022-02-09:
This is a bit confusing to install. I downloaded `FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111.7z` from there https://github.com/realthunder/FreeCAD_assembly3/releases and when i unzip it and start it with `RunFreeCAD.bat` i get the error `<asm3.main> init_gui.py(14): no solver backend found`. 
i found this  https://forum.freecadweb.org/viewtopic.php?f=20&t=57462&start=20 but it was a bit confusing on what was done at all as in the end they copied the asm files into the 'official' ones. 

I have a freecad installation in my system, should this be removed? Should i copy the extraded files into the 'official' freecad directory? `python.exe -m pip install py_slvs` this worked in the 'official' release but not in the realthunder version because of the not bundled python. 

(Or do i need to install python so i can use the python-command?)

luzpaz, 2022-02-09:
[image]

I guess the README presupposes that Window users have python and `pip` pre-installed? (no clue about windows since I'm on linux)

realthunder, 2022-02-09:
There is no need to remove offical FreeCAD installation.

Manually running the pip command can work. But you need to make sure you've `cd` (i.e. change directory command) to the 01.11 release image bin directory (e.g. `C:\Users\<your_username>\Downloads\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\bin`). And run the `Python.exe` there.

But there is a easier way. Starting from the 01.11 release that you are using, Assembly3 workbench can now auto install `SolveSpace` backend solver. If you have not manually cloned the FreeCAD_Assembly3 workbench into your FreeCAD user directory, then with the latest Assembly3 workbench bundled inside the 01.11 release, it will automatically prompt for `SolveSpace` backend installation when you either launch the Assembly3 workbench, or open a file containing any Assembly3 objects.

If you have manaully clone the Assembly3 repository, you can either delete the folder, or `git pull` to update it. 

Note that if after auto install, it still shows some error, try restart FreeCAD.

natrius, 2022-02-09:
Did not consider just trying to start assembly3 :D It did ask indeed, but it showed me an error (i guess from an old pip) but it showed "successfully installed. After restarting everything is fine. The part with 

>If it shows the error `no solver backend found`. Starting from 01.11 release you can open the Assembly3 workbench automatically install the `SolveSpace` backend solver.

Or something like that. Maybe this issue is enough already :) Solved from my part, thanks for your fast response and good work!

closed 2022-02-09 by natrius