realthunder/FreeCAD_assembly3 #1113: Bug: Linux cannot launch appimage Stable nor Daily builds from latest 20220111 Release

eduncan911, 2022-02-10:
I have been using `FreeCAD-asm3-Daily-Conda-Py3-Qt5-20211015-glibc2.12-x86_64` for sometime just fine.  it's working great.

However, neither AppImage for 20220111 is working on my Linux machine.  Running either one does nothing.  No splash, no dmesg or system info, nothing.

Pop_OS 21.10 (aka Ubuntu 21.10)

I've tried with and without my Nvidia card (it's a laptop with both Intel iGPU and Nvidia GPU).

Let me know if there's something else I can try.

realthunder, 2022-02-10:
Please try launching the AppImage in terminal and see if there is any outputs.

eduncan911, 2022-02-13:
Thanks for replying!  Here ya go:

```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage
/tmp/.mount_FreeCAeZIrJy
/tmp/.mount_FreeCAeZIrJy/usr/bin/FreeCADLink: error while loading shared libraries: libOpenGL.so.0: cannot open shared object file: No such file or directory
```

eduncan911, 2022-02-13:
> You are missing the libopengl package. Not sure which library causes the dependency, but it can't be bundled inside AppImage.
> 
> ```
> sudo apt install libopengl0
> ```

Ok, that got it to work.  But this seems like a bigger issue down the road.

I installed a fresh 21.10 in a VM, and tried to the AppImage and had the same error.  

So it seems people with AppImages need to be told to install this going forward?

closed 2022-02-14 by eduncan911