realthunder/FreeCAD_assembly3 #1114: Next Daily/Stable Release tentitive date.

QuakeString, 2022-02-12:
Hi Dev,
I was wandering when next release coming up.
My issue fixed here [0ea6855](https://github.com/realthunder/FreeCAD/commit/0ea68559104c3c285fc5076a5909b74343b21992)
But compiling by myself giving me many error while using FreeCAD.
One of my personal project stuck. Eagerly waiting for next release.

realthunder, 2022-02-12:
If you are using Linux, you can try the [snap build](https://snapcraft.io/freecad-realthunder). If you are on Windows, then you can manually fix the problem by finding the `gui_selectplane.py` file and editing it according to [0ea6855](https://github.com/realthunder/FreeCAD/commit/0ea68559104c3c285fc5076a5909b74343b21992).

QuakeString, 2022-02-12:
Thank You for super quick reply.
I use Manjaro Linux, I'll try snap build now.

closed 2022-04-09 by luzpaz