realthunder/FreeCAD_assembly3 #1115: visibility issue

hamza00789, 2022-02-12:
Sometimes this problem occurs when rolling faces, but it also gets better automatically. 

https://user-images.githubusercontent.com/92737149/153699104-d0b2288a-8dd1-482e-b206-b8ad55b0ca68.mp4 

When I go to a particular view, for example, the top right face sometimes turns black, which automatically recovers when rotated. 

https://user-images.githubusercontent.com/92737149/153699132-225a493a-4948-497f-a9f4-6656f5eced4f.mp4

hamza00789, 2022-02-12:
There is another similar problem.  

A line appears as the sketch moves.

https://user-images.githubusercontent.com/92737149/153729423-199f42d9-77da-446e-b4d4-e018becd6654.mp4

realthunder, 2022-02-13:
I thought you are using the 01.18 release I sent to you? The sketch problem should be fixed there. It is different from the problem in your OP, which is a clipping problem. Either 'View fit' (V, F), or middle mouse click the object to re-center/fit your view.

hamza00789, 2022-02-14:
>It is different from the problem in your OP, which is a clipping problem. Either 'View fit' (V, F), or the middle mouse click the object to re-center/fit your view.  

Doing fit all (V, F) makes it work properly.  
But the shortcut keys stop working. Until the rotation is resumed.  

https://user-images.githubusercontent.com/92737149/153796529-310ad376-6afe-45a1-9c0e-b3291d2d7286.mp4 
 
Maybe this is not a big issue, should I close it?

hamza00789, 2022-02-14:
> I thought you are using the 01.18 release I sent to you?  
No, I'm not using it. You didn't answer whether I should use it or not. I thought maybe I should wait for the new release. I didn't know if this release was for viewing or use.

luzpaz, 2022-04-12:
@hamza00789 can you still reproduce this? (and also try on the release RT sent you?)

hamza00789, 2022-04-13:
> @hamza00789 can you still reproduce this? (and also try on the release RT sent you?)

What answer are you waiting for? Do you want me to re-submit this issue? Should I try again with the new release? Can you explain?

hamza00789, 2022-04-13:
> @hamza00789 can you still reproduce this with the release realthunder sent you? If no, then please notate this ticket so we can close it. 

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)   

This problem is still present in new releases. 

[MP4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8477540/MP4.zip)  

If you want to turn it close, turn it close because that's not a big issue for me.

realthunder, 2022-04-13:
What navigation method are you using? The problem is that your rotation causing the camera to panning out of the scene boundary. You can use mouse middle click the object the recenter rotation center. But I'm curious what causes the panning while rotating.

hamza00789, 2022-04-13:
I am using blender navigation. 

[image]

hamza00789, 2022-04-13:
Looks like this problem is only happening to me. I think the solution you have given is enough.  instead, you need to solve other problems which are more important than that.

closed 2022-04-13 by hamza00789