realthunder/FreeCAD_assembly3 #1117: Part_design-Draft seems to be broken

jj-uk, 2022-02-13:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United Kingdom (en_GB)

12:58:27  <Exception> TopoShapeEx.cpp(1234): Mapped element not found: ;#4b;
12:58:27  Recompute failed! Please check report view.

Create a part - body - sketch on XY plane - centred rectangle.  make top and side equal, set Horz length to 25mm (now fully constrained).
Pad to 10mm (the default when padding).
Select top of cube, click the "Draft" toolbar icon.
Set angle to 20°
Select Neutral plane, click the "front" face (on the left)

At this point, the draft seems to reverse for no reason, so select the "reverse pull direction" checkbox.
Draft is now in the expected correct direction.

Click OK, and you are presented with an error message:

12:58:27  <Exception> TopoShapeEx.cpp(1234): Mapped element not found: ;#4b;
12:58:27  Recompute failed! Please check report view.

TommySchwartz, 2022-03-09:
I also have this wierd "Mapped element not found: ;#xxxx;" error codes. Can't change a dang thing in my design anymore.

jj-uk, 2022-03-09:
> I also have this wierd "Mapped element not found: ;#xxxx;" error codes. Can't change a dang thing in my design anymore.
> 
> Both in macOS and Windows.

Delete the draft.

TommySchwartz, 2022-03-09:
I have about 6 drafts in my design, so I gave up.

realthunder, 2022-03-28:
Same as #1144 and is fixed now

closed 2022-03-28 by realthunder