realthunder/FreeCAD_assembly3 #1119: Spreadsheet: painfully long recompute times, even just changing unaliased text fields

brandonbeaumia, 2022-02-13:
Per this thread in the FreeCAD forums, this bug was sorted out already somewhere in the 0.20 dev branch in early December(ish). Is there any chance we can get the fix merged for next release? On my laptop I'm waiting over 5 seconds between cells just to put some text in, which makes spreadsheets almost unusable.

Unfortunately, I couldn't track down the commit but here's the forum thread where chennes states it is fixed.

https://forum.freecad.org/viewtopic.php?f=8&t=48600&sid=6c48873c06345debe52acdb872e143b8&start=10

realthunder, 2022-02-14:
Can you please post a file showing the problem? If it's sensitive, maybe you can share through my email realthunder.dev at gmail.

brandonbeaumia, 2022-02-14:
[mdt2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8061108/mdt2.zip)

Sorry about that. By the forum post I assumed it was a problem with most/all files. so left it at that. Here's the file (rename .zip to .fcstd)

realthunder, 2022-02-18:
The problem should be fixed with the above commit. Thanks for reporting. Note that there are additional recompute optimization implemented in my branch. The problem here is not related to the provided link.

closed 2022-02-18 by realthunder