realthunder/FreeCAD_assembly3 #1122: Unable to run Appimages on Linux Mint

Tasmedic, 2022-02-15:
Hi there

I have a vanilla install of Linux Mint Mate, as follows
Kernel: 5.4.0-99-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: MATE 1.26.0 
           wm: marco dm: LightDM Distro: Linux Mint 20.3 Una base: Ubuntu 20.04 focal 

I have downloaded the current Daily and Stable appimages and set them as executables but they will not run. Double clicking them does not seem to achieve anything. No error reports, no screen displays... it's as if I haven't done anything at all by clicking on them. 
I'm no linux expert (only running various Debian variants for 10 years or so) but this one has me stuck. 
Are there some additional packages I need to install which I'm unaware of, or didn't notice in the release notes?!

Anyway, I'm grateful for any help, and apologize in advance for missing something simple!

Thanks

Chris.

realthunder, 2022-02-15:
Try run the AppImage from a terminal. Maybe you are missing the libopengl package. Just install it and try again.

Tasmedic, 2022-02-15:
Fixed! Thanks!
That's probably all you need to know, but in case you're interested, please accept my heartfelt thanks for redesigning Freecad so it actually works, and also for your quick reply to my problem. I'm dropping 50 Aussie Dollars into your patreon, in the hope you can help others in the way you've helped me. 
Best wishes
Tasmedic

realthunder, 2022-02-15:
Thank you very much!

closed 2022-02-15 by realthunder