realthunder/FreeCAD_assembly3 #1123: py-slvs bindings not available for Apple Silicone MacBook Pro

shandyba, 2022-02-15:
It appears there are no apple silicone compatible wheels available for `py-slvs` bindings:

```
pip install py-slvs
...
No matching distribution found for py-slvs
```

while running 
```
pip install py-slvs -vvv
```

produces a list of available wheels none of which are apple silicone compatible (`universal` or `arm`), as is expected by observing `pip debug --verbose` output.

Thus, "no solver found" on a MacBook pro with Apple Silicone.

Will appreciate any hints / workarounds / fixes! :)

realthunder, 2022-02-17:
I have just updated py-slvs package with the MacOS universal wheel. Please try again.

shandyba, 2022-02-17:
Hi, thanks a lot, it works now!

closed 2022-02-17 by shandyba